xsp1989/zigbeeFirmware #2: Request for Router FW

tube0013, 2020-09-04:
Would it be possible to get a FW for the bridge to act as a standalone router/repeater?

Hedda, 2021-04-01:
+1 @xsp1989 Any chance you can release a "Zigbee Router" firmware for Zigbee 3.0 USB Dongle Model 9888010100045 ?

https://www.itead.cc/zigbee-3-0-usb-dongle.html

Hedda, 2021-04-01:
It would by the way also be great if we could get a Zigbee Router firmware to Sonoff ZBBridge from ITead if possible(?)

https://www.itead.cc/sonoff-zbbridge.html

Zigbee Router firmware for Sonoff ZBBridge could especially be useful if used in combination with Tasmota on it

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

https://notenoughtech.com/home-automation/flashing-tasmota-on-sonoff-zigbee-bridge/

xsp1989, 2021-04-01:
@Hedda I don't understand what router firmware you are talking about? Is it used as a routing device? Like a onoff light(deviceid:0x1000)

Hedda, 2021-04-01:
@xsp1989 Yes, to use as a Zigbee router (ZR) device for routing, which is sometimes marketed as signal repeater or range extender.

So it would be firmware like any normal mains-powered Zigbee router device, such as a Zigbee power-plug or lightbulb, passing data on from other devices, but the difference here that the Zigbee USB dongle (and Sonoff ZBBridge) would only work as a dedicated Zigbee router and really not control any replays to power-off lights, etc.

Some of the currently popular Zigbee Router alternatives include these:

CC2652P router by tube0013
https://community.home-assistant.io/t/zigbee-router-on-steroids/280896
https://github.com/tube0013/tube_gateways/tree/main/tube_cc_router
https://www.tubeszb.com/product/cc_router/4?cp=true&sa=false&sbp=false&q=false&category_id=3

CC2530 (with CC2591 or CC2592 Power Amplifier) DIY based router as recommended the Zigbee2MQTT community: 
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

IKEA Trådfri Signal Repeater
https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/
https://www.ikea.com/ca/en/p/tradfri-signal-repeater-30400407/
https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

Hedda, 2021-04-01:
Basically; after flashing Zigbee router firmware to Zigbee USB dongle (or Sonoff ZBBridge) you could plug it into a USB-changer to give it power and then pair it to your Zigbee network and then you can place it anywhere within reach to extend range coverage. 

Best it of course to place any Zigbee router device away from but still within reach of your Zigbee coordinator.

As a scenario; if you have a very large house or separate buildings on your property and want to place a battery-powered temperature-sensor far away from your Zigbee coordinator then you will normally need a mains-powered Zigbee device that can act as a Zigbee router which is then placed near to that battery-powered temperature-sensor.

A real-world example where a manufacturer is aware of this range limitation is IKEA who include their "IKEA Trådfri Signal Repeater" packaged with their battery-powered electrical roller blinds which are "FYRTUR" called "KADRILJ" as a combination:

https://www.ikea.com/es/en/p/kadrilj-roller-blind-wireless-battery-operated-grey-20408116/

https://www.ikea.com/us/en/p/fyrtur-blackout-roller-blind-wireless-battery-operated-gray-70417463/

Hedda, 2021-04-01:
Btw, I wrote these sections for Home Assistant's ZHA integration documentation to emphasize the importance of Zigbee routers:

### Using router devices

Zigbee uses mesh networking at the core of its architecture design and adding devices capable of acting as Zigbee router (sometimes also known as Zigbee signal repeaters or extenders) is very important because they increasing the total number of Zigbee devices that your Zigbee coordinator can control in its network as well as having an integral role in extending the overall range and coverage of your Zigbee network mesh. The fact is that without having at least a couple of good Zigbee router devices available you will normally not have a stable Zigbee network.

The total number of Zigbee devices that you have on a Zigbee network depends on a few things, but you should know that Zigbee coordinator hardware and firmware usually only start to play a larger role once you have around 40 devices or more in your Zigbee mesh network. More important is how many directly connected devices ("direct children") versus how many routers are and can be connected to your Zigbee coordinator. Each type and model of Zigbee coordinator also gave a different limit for how many child devices can connect directly to it. Even if that limit is low on your Zigbee coordinator if your Zigbee coordinator hardware is powerful and its firmware capable of it then you can still have a total of hundreds or even thousands of Zigbee devices as long as they are indirectly connected through a mesh network with Zigbee router devices instead.

The least powerful Zigbee coordinator hardware supported by Zigpy is CC2530/2531 and its default firmware only supports 15 devices connected directly to the coordinator. However, by having a backbone of good always-on Zigbee routers in your Zigbee network, the mesh network size can be extended exponentially. You can assume that most, if not all mains-powered/AC-powered devices, e.g., wall-plugs and always powered-on lightbulbs in your Zigbee network can serve as a Zigbee router and can typically each act as a router for around 5 additional end devices, while some dedicated Zigbee router devices can each act as a router for as many as 50 devices. You can for example use a DIY CC2530/CC2531 with router firmware which have a limit of 21 devices, while a DIY CC2652P/CC2652R with router firmware which have a limit of 50 devices, or the IKEA TRÅDFRI Signal Repeater which can route about 10-20 devices.

An example using a CC2531 with default coordinator firmware + two CC2531 routers; Your Zigbee network device limit will be:

- Coordinator: 15 devices - 2 routers = 13
- Router one: + 21 devices
- Router two: + 21 devices
- Total device limit = **55 devices**

As an comparison example, using a CC2652 with default coordinator firmware + two CC652 routers; Your Zigbee network device limit will be:

- Coordinator: 50 devices - 2 routers = 48
- Router one: + 50 devices
- Router two: + 50 devices
- Total device limit = **148 devices**

### Improving Zigbee network range

Low signal quality can lead to transmission errors and related issues. This section has a list of fundamental tips on how to improve signal quality. Improving signal quality also maximizes range and resolves most problems related to transmission errors. Please try to follow at least some of these recommendations before posting on the community forums or reporting issues to developers and submitting debug logs.

1. **Add more Zigbee routers between devices far away and the next closest router. Distribute more Zigbee routers in areas with poor reception. Zigbee network topology uses a "mesh network" design which means that each device that acts as a Zigbee router will extend the range of your Zigbee network as a whole. While there are exceptions, almost all permanently powered devices serve as a Zigbee router. Adding more permanently powered Zigbee devices allows getting better coverage. There are also dedicated Zigbee routers which you can [find by doing a community search for "Zigbee signal repeater" or "Zigbee range extender"](https://community.home-assistant.io/t/advice-on-zigbee-range-extending/175882/)). Devices that can not act as routers are typically battery-operated and known as "end devices". Some end devices have issues connecting through routers (e.g. Xiaomi/Aqara devices).**
2. Use a USB extension cable. This allows positioning the Zigbee coordinator adapter for better signal quality. Position the Zigbee coordinator away from Wi-Fi access-points/routers or other sources of 2.4GHz signals to avoid signal interference. The best location depends on your building's floorplan. Ideally, you want to place the Zigbee coordinator somewhere in the middle. Building materials do influence signal quality too, for example, dense/thick concrete, bricks, masonry, etc. dampen all wireless signals. Place the Zigbee coordinator with some distance from walls, ceilings and floors. Also, try different orientations of your Zigbee antenna (or your whole Zigbee coordinator adapter if it has an internal antenna). Some Antennas have a stronger signal in a certain direction. Simply changing orientation can improve signal quality already. Note: A bad USB extension cable may lead to connection issues between the system and the Zigbee coordinator adapter, symptoms of this are disconnection messages.
[... cut ...] , ceilings and floors.
6. Buy more powerful Zigbee radio hardware with better radio range, preferably with an external antenna. If you are not only experimenting but want a permanently stable and healthy Zigbee network with potentially many devices then you should consider upgrading to a more powerful Zigbee coordinator USB adapter. Generally, those with an external antenna will have better range, therefore you will also want to avoid buying an internal adapter unless it has an external antenna.

Hedda, 2021-04-01:
Just to clarify; this "Zigbee Router" firmware would be an optional alternative to the existing "Zigbee Coordinator" firmware.

You obviously can not have "Zigbee Router" firmware and "Zigbee Coordinator" firmware flashed at the same time.

People would have to buy at least one additional device if they want to have the same type of devices as a router as well.

Zigbee Coordinator to RaspPi <-> Zigbee Router (Zigbee USB with router firmware) <-> Zigbee End Device (battery-operated).

Hedda, 2021-04-01:
> Is it used as a routing device? **Like a onoff light(deviceid:0x1000)**

@xsp1989  Please note that we of course would it capable of full Zigbee 3.0 (Z3) compliant Zigbee Router.

So not only routing for Zigbee Light Link (ZLL) or only or Zigbee Home Automation (ZHA) devices.

Hedda, 2021-04-01:
One of the buttons on the Zigbee USB stick would obviously have to be mapped to activate pairing mode on the device.

xsp1989, 2021-04-01:
@Hedda If you have Simplicity Studio, you can try to develop your own routing firmware. Sonoff ZIGBEE 3.0 USB DONGLE firmware does not require signatures.

xsp1989, 2021-04-01:
If you compile the firmware yourself, please set the HFXO CTUNE value to 128.

Hedda, 2021-04-01:
Sorry but I have not yet bought an official development kit which I understand is still required to get access to their Zigbee stack(?).

What is your opinion on router firmware for Sonoff ZBBridge, see a separate discussion about that here -> https://github.com/arendst/Tasmota/discussions/11536

Would you be willing to compile a signed "Zigbee Router" firmware for ITead's Sonoff ZBBridge for Tasmota to enable pairing mode?

Hedda, 2021-05-28:
FYI, @MattWestb made a hardware configuration of a Zigbee router firmware made for a Silabs EmberZNet EFR32 module:

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/issues/6

"Router firmware with pins for E1743 remote (paring / reset button = pairing button and also BLB if using my bootlooader)."

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/files/6559440/ZigbeeBillyRouter.zip

https://github.com/zha-ng/EZSP-Firmware/files/6560171/ZigbeeBillyRouter.zip

Hedda, 2021-05-28:
By the way, Tuya is apparently making a 10 dBm "Zigbee 3.0 Range Extender" / "Zigbee 3.0 Signal Repeater" (Tuya SKU: "RP280") described as Zigbee Router device in a USB dongle/stick format -> https://zigbee.blakadder.com/Tuya_TS0207.html (Zigbee 3.0 Signal Repeater Model TS0207 manufactured by Tuya. Zigbee ID: "TS0207"; "_TZ3000_m0vaazab"; "FNB54-WTS08ML1.0").

Unfortunately, this specific Tuya RP280 USB Zigbee router been proven to cause network stability issues. read -> https://github.com/zigpy/zigpy/discussions/736

The concept is nice though.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

xsp1989, 2021-12-06:
This is the routing firmware：
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/RouterForDongle.gbl

Hedda, 2021-12-06:
Cool! Which exact EmberZNet version is it based on and is any of the buttons mapped as a paring button to enable join?

Maybe perhaps detail the steps for putting into joining mode? And steps to reset, etc.

xsp1989, 2021-12-07:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle/RouterForDongle

@Hedda

Hedda, 2021-12-07:
So version EZSP 6.7.10? ...sorry, version used is still not 100% clear when the version is not part of the filename.

@xsp1989 Suggest/recommend rename "**_RouterForDongle.gbl_**" to "**_efr32mg21_zigbee_router_6.7.10.gbl_**" or something similar.

Version in filename is also important if ever release another version.

@xsp1989 Any chance you could add patch or config info to readme so others can build and compile their own router firmware?

xsp1989, 2021-12-07:
@Hedda The file name has been changed

Hedda, 2021-12-08:
@xsp1989 Any chance you could provide ITead signed router firmware for Sonoff ZBBridge? It too uses an SM-011 module.

For reference; It is currently popular to flash ITead's Sonoff ZBBridge with either Tasmota or ESPHome and use it from [Home Assistant's ZHA integration](https://www.home-assistant.io/integrations/zha) as a remote Zigbee Coordinator (via EZSP) using a serial stream server in Tasmota/ESPHome, but since ITead's Sonoff ZBBridge has been shown to be unstable as such reote Zigbee Coordinator in ZHA because of its WiFi-only connection to the network it would be great if could instead use it as a Zigbee Router with Zigbee Router firmware instead, however, I believe it requires firmware image for ITead's Sonoff ZBBridge to be signed to work?

https://zigbee.blakadder.com/Sonoff_ZBBridge.html

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

https://tasmota.github.io/docs/Zigbee/

https://github.com/thegroove/esphome-zbbridge/

PS: Alternative to ITead's Sonoff ZBBridge hardware is ZB-GW03 eWeLink Ethernet LAN Zigbee Gateway as also uses the SM-011:

https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-zbbridge-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-adapter-with-home-assistant-zha/341223/26

https://templates.blakadder.com/ewelink_ZB-GW03

https://thehelpfulidiot.com/a-wired-sonoff-zigbee-alternative

https://github.com/syssi/esphome-zb-gw03

xsp1989, 2021-12-08:
There is already a signed routing firmware for ZBbridge, and it may not be released. If you need it, I can send it to you by email.

digiblur, 2021-12-08:
> There is already a signed routing firmware for ZBbridge, and it may not be released. If you need it, I can send it to you by email.

I'd be curious to see how it works.  I know I've had several people asking about routing firmware on the ZBBridge to repurpose them.

Hedda, 2021-12-08:
> > There is already a signed routing firmware for ZBbridge, and it may not be released. If you need it, I can send it to you by email.
> 
> I'd be curious to see how it works. I know I've had several people asking about routing firmware on the ZBBridge to repurpose them.

Would especially be interesting if could control pairing/joining it as a Zigbee device from Tasmota and/or ESPHome builds for it.

MattWestb, 2021-12-08:
I think the ZBB dont have the GPIO connected to the pad of the Zigbee module but perhaps its working using the reset pad of the module if the code have logick implanted for it (like most lights is having).
The serial is used for the tunneling and i think its not possible sending any commands to the Zigbee stack thru it.

xsp1989, 2021-12-15:
ZBB needs to develop functions on ESP8266, pull down PA00 to enter configuration mode .

xsp1989, 2022-01-26:
> > There is already a signed routing firmware for ZBbridge, and it may not be released. If you need it, I can send it to you by email.
> 
> I'd be curious to see how it works. I know I've had several people asking about routing firmware on the ZBBridge to repurpose them.

The routing firmware is not specially adapted to ZB-Bridge, which is a commercial firmware controlled by a serial port protocol. Update files can be released if needed, but it's not a good idea to use this firmware on ZBB

digiblur, 2022-01-26:
I think it would be great as just a dumb router per say. As people have these laying around in the drawer and could write the router firmware to it and join it to their main zigbee network. The bridge via serial wouldn't do anything of course.

Hedda, 2022-01-26:
> The routing firmware is not specially adapted to ZB-Bridge, which is a commercial firmware, controlled by serial port protocol, or the same way as 3.0 dongle (PA00 is connected to GND for 5 seconds). Update files can be released if needed, but it's not a good idea to use this firmware on ZBB

@xsp1989 It is already very popular to flash ITead's Sonoff ZBBridge with Tasmota or ESPHome firmware to use a coordinator.

I am sure that if a Zigbee Router firmware was released for it then it too would also become popular to use as a dedicated Zigbee Router among the same communities that use ITead's Sonoff ZBBridge with Tasmota or ESPHome.

One benefit of having access to it via Tasmota or ESPHome would be to remotely reset/reboot it if hangs for whatever reason.

I strongly believe that ITead would actually sell more Sonoff ZBBridge devices to those community members if a compatible Zigbee Router firmware was released for it.

Hedda, 2022-01-26:
@xsp1989 By the way, check out these related blog posts by @digiblur which are good examples of how news like this is spread:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

digiblur, 2022-01-26:
@Hedda the reuse of hardware is a beautiful thing instead of ending up in a landfill.

Hedda, 2022-01-27:
> @Hedda the reuse of hardware is a beautiful thing instead of ending up in a landfill.

Yes I agree, though my main argument is still it would bring new customs to buy multiple Sonoff ZBBridge just to use as routers.

They can practically only sell one Zigbee Coordinator to most customers but they can sell several as routers to each customer.

Sonoff ZBBridge and ZB-GW03 has both a very nice form factor and many users like the idea of Tasmota or ESPHome enabled.

It could work as a Tasmota and ESPHome administration enabled alternative to Tube's Zigbee Router and IKEA Signal Repeater:

https://www.tubeszb.com/product/cc_router/4?cp=true&sa=false&sbp=false&q=false&category_id=3

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/

https://www.ikea.com/se/sv/p/tradfri-signalfoerstaerkare-10400408/

Hedda, 2022-02-17:
FYI, @xsp1989 has uploaded/shared a signed Zigbee Router firmware for ITead's Sonoff ZBBridge to his google drive here:

> https://drive.google.com/file/d/1H-M5CTh_XGGVl2te4vHLQ5SfGPHL6yCS/view?usp=sharing
> 
> This is the signed routing firmware used by ZBB, the usage method is the same as the unsigned firmware.
> 
> https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md

PS: This idea also discussed specifically for ITead's Sonoff ZBBridge hacked with Tasmota here -> https://github.com/arendst/Tasmota/discussions/11536

Hedda, 2022-02-18:
Update from xsp1989 on router firmware posted on https://github.com/arendst/Tasmota/discussions/11536

"This firmware can ground the PA00 5S into pairing mode, and ground the RST to restart routing. UART is used to transmit data transparently.

and

"There is also another version of the routing firmware that can control the join network through a proprietary serial port protocol"

Hedda, 2022-02-20:
Check out this new "How to Convert the Sonoff Zigbee Bridge into a Router / Repeater" article by @digiblur here:

https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html

Note that the article already assumes that you have flashed Tasmota onto ITead Sonoff ZBBridge as per this guide:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

joe-sydney, 2022-07-21:
I followed the digiblur instructions and was able to convert my Sonoff Zigbee Bridge to a router. Works well, except I notice it never had more than 6 direct children. I didn't expect this as the zigbee bridge can handle many more child devices as a coordinator. Am I doing something wrong?

Hedda, 2022-08-03:
> I followed the digiblur instructions and was able to convert my Sonoff Zigbee Bridge to a router. Works well, except I notice it never had more than 6 direct children. I didn't expect this as the zigbee bridge can handle many more child devices as a coordinator. Am I doing something wrong?

Which Zigbee gateway solution are you using and can you try enabling "Zigbee source routing" in that Zigbee gateway solution?

https://www.digi.com/resources/documentation/Digidocs/90002002/Concepts/c_zb_source_routing.htm?TocPath=Transmission%2C%20addressing%2C%20and%20routing%7CRF%20packet%20routing%7CSource%20routing%7C_____0

If you have Zigbee devices that the manufacturer has properly configured the firmware then they will usually around every 24 hours or so try to connect to the Zigbee Router with the best RSSI (Received Signal Strength) then it will over time report LQI (Link Quality Indication). If you have  Zigbee devices that the manufacturer has properly configured the firmware then this should have no problems routing 20 devices.

But note that if you are using cheap Zigbee end devices from Aqara/Xiaomi or Tuya then know that they commonly do not have properly configured firmware and have a tendency to not connect to other Zigbee Routers from other brands. Some Zigbee solutions do however offer the option to force the initial pairing/joining via a specific Zigbee Router and that will make the device stick to it, though the downside to that is that such devices will then normally automatically switch to another Zigbee Router if the device it is paired through goes down for whatever reason.

Obviously no devices will connect to a Zigbee Router if they see better RSSI from your Zigbee Coordinator and it has available slots.

xsp1989, 2022-09-05:
Routing firmware has been released, so close.

closed 2022-09-05 by xsp1989