xsp1989/zigbeeFirmware #3: Firmware update request for SM-011.

MattWestb, 2021-01-17:
Sonoff Zigbee bridge is having problems setting up pull control on devices that resulting battery in sleeping end devices is draining the battery very fast.

Silabs have finding one bug that is more than likely is doing the the SM-011 modul on the Sonoff Zigbee Bridge.

From latest release of [Zigbee EmberZNet SDK 6.7.8.0 GA](https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.8.0.pdf) :

`638989` `Fixed an issue in the host-ncp sleepy end device configuration, where it continued short polling if the parent connectivity was lost temporarily during data transfer.`

The bug is fixed in EZSP 6.7.8.0 and **not** in 6.8.x or 6.9.x so only the specific version is helping getting the systems working normal.

If its possible for you making one upgraded firmware signed with the certificate for Sonoff Zigbee Bridge its helping our users solving the problems that is making there controlling device draining the batteries.

Thanks in advance and hopeng on more success with the Sonoff zigbee products.

Mattias W

xsp1989, 2021-01-19:
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/SM-011/ncp-uart-sw_6.7.8_115200.ota

I have updated the ncp firmware。Please test it

MattWestb, 2021-01-19:
Thank you very much we will twisting it and reporting back how its working.

Regards,

Mattias W

MattWestb, 2021-03-04:
One more time thanks for the updated firmware.

And i was forgetting to closing the first case.

Regards 

Mattias W

closed 2021-03-04 by MattWestb

MattWestb, 2021-04-01:
Thanks @silabs-RaoulvB  I was looking thru 6.8.X and was not see it in it and it was older then the 6.7.8.0 that was flagged as fixed so i was expecting the fix was not implanted in 6.8.x.

Thanks for clarifying my misunderstanding  !!