xsp1989/zigbeeFirmware #4: [BUG] Missed to set HFXO CTUNE value to 128 in firmware for ITead Zigbee 3.0 USB Dongle?

Hedda, 2021-04-12:
@xsp1989 Not sure if you have already heard or not but it seems that a lot of people who bought the new ITead Zigbee 3.0 USB Dongle are having problems when trying to pair their Zigbee end devices to the dongle/stick adapter as a Zigbee coordinator.

See bug reports in -> https://github.com/home-assistant/core/issues/48592

The problem doesn't sound to be isolated to ZHA in Home Assistant but also to OpenHAB users (and Zigbee2MQTT dev users) too. 

@MattWestb in https://github.com/zigpy/zigpy/discussions/692 belive that the root cause is "CTune value" not being tuned in firmware(?).

> Latest new is that the firmware maker is:
> 
> > **_If you compile the firmware yourself, please set the HFXO CTUNE value to 128._**
> 
> So my theory is confirmed !!!!
> The SM-011 and the USB stick is not calibrated or they have changing components that is miss tuning the RF radio !!!

@xsp1989 If so then the solution might be to create a new firmware with HFXO CTUNE value set to 128 or what do you think?

> some one that have on SM-011 module and and Simplicity Studio can reading the CTune value in the chip and /or in the manufacturing token ?
> 
> One KB how is being done in SS4: https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2019/01/22/saving_ctune_valuea-qCPl
> 
> I suspecting that the SM-011 module and the new Sonoff USB Zigbee 3 stick is **not** calibrated and have problems with staying on the correct frequency and reading CTune can saying if they have storing somthing or not in the chip. And the manufacture is trying patching the config in the EZSP firmware for compensating the bad done work.
> 
> The procedure for doing on real calibrating if some is interested:
> https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2019/03/18/hfxo_capacitor_bank-7uRt
> 
> If my thinking its true it can resulting in that devices (sleeping end device mostly) is heaving problem communicating with the coordinator then its frequency is miss tuned or need re tuning its radio all the time for keeping the radio channel open.
> 
> If its possible with one factors firmware that is signed and the chip is not erased and not with EZSP 6.7.9.0 that have some "fixed configuration" but all reading / info is more then welcome !!
> 
> The Silabs modules IKEA and Tube is using shall being OK from the factory if i have understanding all OK but i dont now its with tuya having doing it OK or not on there EF32MG2X modules.
> 
> Edit: One more interesting thing it can being bad using high power setting then it can doing the the RF drifting of the heating in the AP and getting problems with come devices that is tuned in the other direction ( on + x and the other - y) and getting problems.

xsp1989, 2021-04-12:
In version 6.7.9, the CTUNE value has been set to 128

Hedda, 2021-04-15:
Is setting CTUNE value 128 the permanent setting for all Zigbee 3.0 USB Dongle Model 9888010100045 with hardware version 1.3?

https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2019/03/18/hfxo_capacitor_bank-7uRt

Hedda, 2021-04-15:
Does the LFXO Capacitor Bank (CMU clock) also have to be calibrated and corrent tuning value set for that as well in the firmware?

https://www.silabs.com/community/mcu/32-bit/forum.topic.html/clibrate_lfxo_inefr-3I0T

https://www.silabs.com/community/mcu/32-bit/forum.topic.html/how_do_i_use_thetun-Pqlx

https://www.silabs.com/documents/public/application-notes/an0004.2-efr32-series2-cmu.pdf

https://www.silabs.com/documents/public/application-notes/an0004.1-efm32-cmu.pdf

"_This KBA focuses only to HFXO calibration **but the LFXO has its own capacitor bank.**_"

https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2019/03/18/hfxo_capacitor_bank-7uRt

xsp1989, 2021-04-16:
HFXO CTUNE is set to 128 to have better RF performance, and the default value can also work normally

closed 2021-04-21 by xsp1989

Hedda, 2021-05-06:
Sounds as if the root cause for the problem with pairing due to Electromagnetic interference (EMI) / radio-frequency interference (RFI) https://github.com/home-assistant/core/issues/48592 issues is still related to a hardware design flaw as I now read that EFR32 with a properly designed circuit board design for a PCB trace antenna hardware should not need to change the HFXO CTUNE from the default parameter.

It sounds more and more like ITead engineers needs to take this board back to the drawing board and redesign the antenna circuits. As noted before, the ITead Zigbee 3.0 USB Dongle Model 9888010100045 currently do not follow Silabs reference antenna design.

Hedda, 2021-05-06:
Nice blog post about ceramic antennas: 

https://resources.pcb.cadence.com/blog/2020-understanding-ceramic-chip-antenna-vs-pcb-trace-antenna

Technical description:

 https://www.mouser.co.id/pdfDocs/ceramicchipantennasvspcbtraceantennasacomparison.pdf

Ceramic chip antenna is probably worth the ~ $1 that it would add to total BOM cost, even if raising retail price of the product.

https://eu.mouser.com/Passive-Components/Antennas/_/N-8w0fa?P=1y9hq54Z1yu8mv5

MattWestb, 2021-05-06:
I agree but some extra capacitors and shield for the for the HFXO / receiver path is higher on the list for getting it working more stable.

Also possible potting the output power down can helping if the interference is coming from the RF part that is blocking the device.

I was suggesting putting some isolating tape and then aluminium / copper tape that is not covering  the antenna and grounding it at the USB but no one have trying it what i knowing but its one well proven method in SDR dongles community.

1$ in BOM is in the end more § in the bill to the customer so i think one plastic tube / film with metal / shielding coating is one more cost efficient solution in on low cost device like this.

Hedda, 2021-05-06:
> Also possible potting the output power down can helping if the interference is coming from the RF part that is blocking the device.

FYI, such workarounds are very unpopular at least and can have an impact on a company's reputation.

It is not ethical and maybe not even legal to first release a product with "20dBm" listed in its specification and then later say that due to a hardware design flaw you can now only use the product you bought with example only a "12dBm" configuration.

While not as costly you can compare such a solution to Apple's iPhone battery operating "fix" where they just sent out a software patch that lowered the processor frequency so that it would draw less power.

That would also be like Tesla releasing a new "Long Range" model car that they market to have a 400 miles range but then they discover that it has problems so they later release an over-the-air update that limits the car to 300 miles range.

MattWestb, 2021-05-06:
And shipping devices that not working at all under "normal conditions" is . . .  ?
By the  way is one DIY device and its your responsibility doing the casing and connection of it by the recommendations of the  manufacture.