xsp1989/zigbeeFirmware #10: [QUESTION] Calibrate and tune LFXO Capacitor Bank (CMU clock) in firmware for ITead Zigbee 3.0 USB Dongle Model 9888010100045 with hardware v1.3?

Hedda, 2021-04-15:
Note! This is a follow up to https://github.com/xsp1989/zigbeeFirmware/issues/4 as a separate issue so the other can be closed separately when resolved.

Does the LFXO Capacitor Bank (CMU clock) also have to be calibrated and the current tuning value set for that as well in the firmware for ITead Zigbee 3.0 USB Dongle Model 9888010100045 with hardware version 1.3 or?

https://www.silabs.com/community/mcu/32-bit/forum.topic.html/clibrate_lfxo_inefr-3I0T

https://www.silabs.com/community/mcu/32-bit/forum.topic.html/how_do_i_use_thetun-Pqlx

https://www.silabs.com/documents/public/application-notes/an0004.2-efr32-series2-cmu.pdf

https://www.silabs.com/documents/public/application-notes/an0004.1-efm32-cmu.pdf

"This KBA focuses only to HFXO calibration but the LFXO has its own capacitor bank."

https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2019/03/18/hfxo_capacitor_bank-7uRt

Hedda, 2021-04-15:
Might this also be related to issues reported in https://github.com/home-assistant/core/issues/48592 ?

xsp1989, 2021-04-15:
The firmware does not use LFXO （low frequency crystal oscillator）and does not require calibration。

closed 2021-04-15 by xsp1989