xsp1989/zigbeeFirmware #16: [REQUEST] Firmware image pre-configured to make ITead Zigbee 3.0 USB Dongle into a dedicated stand-alone Zigbee Router device?

Hedda, 2021-04-23:
@xsp1989 @silabs-RaoulvB How about a separate firmware image pre-configured to make it a stand-alone Zigbee Router device?

The purpose of such a separate pre-configured firmware image would so that any user can flash it to ITead Zigbee 3.0 USB Dongle and that would by default without any more configuration be optimized to act as a dedicated stand-alone Zigbee Router device.

The point of such a pre-configured firmware image is so the users can buy additional ITead Zigbee 3.0 USB Dongles and togther with USB-charger make those into stand-alone Zigbee Router devices that work similarly to the "**IKEA Trådfri Signal Repeater E1746**" product, meaning that that can be paired by simply enabling paring-mode using just a button pressing combination.

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/
https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

That is, with an optional firmware that is already pre-configured as a stand-alone Zigbee Router all the end-user needs to do is:

1. Flash the pre-configured "**Stand-alone Zigbee Router Device**" firmware image to the ITead Zigbee 3.0 USB Dongle.
2. Plug-in the ITead Zigbee 3.0 USB Dongle into any USB charger.
3. Initiate paring-mode by quickly pressing the "boot" button three times (3 times within a time frame of 3 seconds).
4. LED light blink constantly to indicate that it has initiated pairing-mode (for ~ 10-second before it ends pairing-mode).
5. If the paring is successful then LED light stays in ON mode to indicate that the device is paired and is powered on.

To reset the device to its "default settings" 

1. Power-cycle it (by either pressing "reset" button or by unplugging the device from the USB charger and replugging back in).
2. Within 60-seconds after power-on; quickly press "boot" button six times (6 times within time frame of 3 seconds after powered).

[image]

silabs-RaoulvB, 2021-05-03:
Hi Hedda,
this will only work with touchlink, like IKEA uses, and with IKEA selling a repeater with USB charger for €10, i do not see the point here. https://www.ikea.com/de/de/p/tradfri-signalverstaerker-10400408/
if you want to be able to add the stick as a router in Zigbee 3.0 networks with (3.0) security, you can do it with the ncp firmware and giving it the right commands. Someone just needs to write some software to do this.
regards, Raoul

MattWestb, 2021-05-03:
Great point SL Raoul !!

I have rolling one of this for IKEA ICC-A-1 module but not testing it.

ZB device type: **HA range extender**.
And adding the plugin: **Button form/join code**
>(Requires Network Creator Security + Network Creator or Network Steering plugin to function properly in Zigbee 3.0 networks.  Else, Network Find plugin is strongly recommended but not mandatory.)  This code will hook up button 0 to have specific behavior based on the current network state.  The behavior is as follows. If the device is not joined to a network, it will form a Zigbee 3.0 network (via Network Creator plugin) if it is configured as a coordinator or join a Zigbee 3.0 or ZHA legacy network (via Network Steering plugin) if it is configured as a router. If the device is joined to a network then pressing the button will broadcast a ZDO permit join to allow new devices to join.  Holding the button for 5 seconds and releasing will cause the device to leave the network.  Button 1 is not used and a callback is provided to another module wishing to use it.

Putting the bootloader pin as joining button and you is on the train only 130€ lesser and (included one WSTK and one thunderboard scene 2) and you can getting  13 for that price from IKEA.

The large problem is that the thunderboard is not supporting hardware flow control so its no go for some one (if hot soldering some extra cables).

And with the IKEA you is getting on working PG proxy server that can being implanted in the above firmware 2 but i cant testing if its working.

silabs-RaoulvB, 2021-05-04:
> The large problem is that the thunderboard is not supporting hardware flow control so its no go for some one (if hot soldering some extra cables).

You are tallking here about the Thunderboard Sense 2? It definitely supports HW flowcontrol, maybe I misunderstand this sentence?

Hedda, 2021-05-04:
> this will only work with touchlink

Why could it only work with touchlink? Why could not add to the network via the Zigbee coordinator or a other Zigbee router?

> with IKEA selling a repeater with USB charger for €10, i do not see the point here. https://www.ikea.com/de/de/p/tradfri-signalverstaerker-10400408/

In my experience, if you for example are a new Zigbee user in a house that is either large or built from signal killing materials who wants to begin adding a bunch of wireless sensors in the format of inexpensive battery-operated Zigbee end devices then directly after you setup your Zigbee coordinator you really want to start out by first adding at least three Zigbee router devices spread out in your house to get a somewhat stable baseline network mesh going before you even think about adding any of those battery-operated wireless Zigbee end devices around your house.

Therefore, if using these ITead Zigbee 3.0 USB dongles as routers in USB-chargers was possible in that scenario, the least expensive option today to get started would be to buy four of these ITead Zigbee 3.0 USB dongles for a total of $21 and then use one of the dongles as the Zigbee coordinator in a Raspberry Pi and the other three dongles as Zigbee routers in the combination with USB-chargers that you already one.

Going that way you saved at least $20 compared to buying IKEA's Tradfri Signal Repeater products, and at least an additional $3 for each additional such router if you only plan on adding wireless sensors and just need mains-powered devices to function as routers.

MattWestb, 2021-05-04:
@silabs-RaoulvB Yes its written little strange but looking on the schematics is the J-tag / USB comport supporting HW but if using the mini connector its dont supporting it (no problem for my then using SW) but its still problem getting vcom and PTI working on WSTK in out mode to external boards like Thunderboard and custom one.

I still dont understand way like using one "stripped " USB stick as on router then its getting ready to go product on the market but perhaps shall we dont understanding all things in the world.

silabs-RaoulvB, 2021-05-05:
Hi Hedda,
I understand and respect your reasoning, as it certainly is valid for you.
Please also accept my reasoning, that it is not for me. First I do not have so many usb chargers laying around, second my wife would never accept such ugly repeaters, third I installed light bulbs in strategical places as repeaters and enjoy the added value and comfort of those as well.
So sorry, hope you understand I will not invest many hours of my precious time generating a firmware I have no benefit from. Maybe you are lucky and find someone who will.

My remark on touchlink was based on the fact that any push button joining method (so without install code) is considered a safety threat, so I expect these to disappear over time.

take care.

Hedda, 2021-05-11:
FYI, apparently, Tuya has seen the need for such a product and are now making and selling Tuya Zigbee 3.0 USB Range Extenders.

Tuya SKU: RP280

It is however not clear if it really works as a Zigbee Router with third-party gateways or if need Tuya gateway as their description:

https://community.home-assistant.io/t/tuya-zigbee-signal-repeater-working-with-zha-and-conbee-2/249329

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4175

https://github.com/Koenkk/zigbee2mqtt/issues/4104

https://zigbee.blakadder.com/Tuya_TS0207.html

Signal Repeater Zigbee 3.0 Model TS0207 manufactured by Tuya

Zigbee ID: "TS0207"; "_TZ3000_m0vaazab"; "FNB54-WTS08ML1.0"

https://www.aliexpress.com/i/1005001459823340.html?spm=2114.12057483.top.4.55932e050QWoD7

https://www.aliexpress.com/i/1005001274714963.html

https://www.aliexpress.com/item/1005002441359324.html

https://www.aliexpress.com/i/1005002526017479.html

https://www.amazon.com/Extender-Protable-Amplifier-Gateway-Repeater/dp/B08MWBKK4P

https://www.amazon.com/Repeaters%EF%BC%8CKKmoon-Amplification-Transmitters-Enhancement-Compatible/dp/B08R3QC9X6/

[image]

[image]

[image]

[image]

[image]

[image]

[image]

MattWestb, 2021-05-11:
I wondering id they is having one "compact PCB antenna" and how good the RF part is working in the real.
 Its more or less the format as ELibs USB dongle but i think they dont have any USB-serial chip on it then its not needed.

I have only seen some user using it and the LIQ was not one high light but i think what you can expecting of the device.

Andy internal photos of FCC link ??

Chip CC-2530 ?

Hedda, 2021-05-28:
FYI, MattWestb made a hardware configuration of a Zigbee router firmware made for a Siicon Labs EmberZNet EFR32 module:

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/issues/6

"Router firmware with pins for E1743 remote (paring / reset button = pairing button and also BLB if using my bootlooader)."

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/files/6559440/ZigbeeBillyRouter.zip

https://github.com/zha-ng/EZSP-Firmware/files/6560171/ZigbeeBillyRouter.zip

Config made for Silabs modules ripped from IKEA Trådfri devices https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/

PS: To clarify; this is only useful for building and compiling your own firmware so you can not just flash that as a firmware image.

Hedda, 2021-08-24:
FYI, looks like MattWestb has now built/compiled a Zigbee Router test firmware made for Silicon Labs EmberZNet EFR32 modules:

https://github.com/MattWestb/EFR32-FW/tree/main/Soff_EZSP

https://github.com/MattWestb/EFR32-FW/tree/main/Soff_EZSP#soff-ezsp-router

@MattWestb Do I understand correctly that a module with "Soff EZSP Router" firmware can be used as both Zigbee Coordinator or a Zigbee Router depending on the state of the Zigbee network when it is joined? 

That is, a dongle with your "Soff EZSP Router" firmware could be used as a Zigbee Coordinator if you just use it like any other EZSP Zigbee Coordinator, otherwise, it could be joined to an existing already formed Zigbee network as a Zigbee Router by pressing the bootloader button to enable join mode when it is already powered on?

Anyway, he noted that this firmware has not been fully tested but I understand it should work on the ITead Zigbee 3.0 USB Dongle(?):

### Soff EZSP Router
  
Configuration Parameter | Value
-- | --
Module | SM-011
Part | EFR32MG21A020F768IM32
Version | EZSP 6.10.1.0
Status |  Untested
CTUNE value | 128
TX | PB01
RX | PB00
BTL | PA00 (Low = BTL boot)
NA | PA01 (dummy for compilling the plugin)
LED | PC00 
    
> From the pluging description:
> > (Requires Network Creator Security + Network Creator or Network Steering plugin to function properly in Zigbee 3.0 networks. Else, Network Find plugin is strongly recommended but not mandatory.) This code will hook up button 0 to have specific behavior based on the current network state. The behavior is as follows. If the device is not joined to a network, it will form a Zigbee 3.0 network (via Network Creator plugin) if it is configured as a coordinator or join a Zigbee 3.0 or ZHA legacy network (via Network Steering plugin) if it is configured as a router. If the device is joined to a network then pressing the button will broadcast a ZDO permit join to allow new devices to join. Holding the button for 5 seconds and releasing will cause the device to leave the network. Button 1 is not used and a callback is provided to another module wishing to use it.  
>
>Compleet CLI with help is in the firmware for debug and also joining / leaving the network from CLI.
I recomending doing one chip erase and flashing one bootlooader and then the router S37 firmware but i attaticg the GBL if like trying doing it without erasing the flash.  
Pins is taken from the USB Stick that have the same pins plus one LED and shall working the same.  
The joining pin is the same as being used for bootloader so possible using it in Tasmota or ESPHome as switches.  

PS: Off-topic for this issue but also interesting that his firmware build is based on EmberZNet 6.10.0.0 (more info about that in https://github.com/xsp1989/zigbeeFirmware/issues/18).

MattWestb, 2021-08-24:
The router firmware is EZSP 6.10.1.0 and is device type HA range extender so it cant being used as coordinator only router (its not allowed adding extra function in this profile if reading the Zigbee documents also no GP proxy). 
As it not signed it shall working (if its working at all) on ZBB with flashed bootloader, USB Stick and the new ESP32 Eth Zigbee bridge.

Hedda, 2021-08-24:
> The router firmware is EZSP 6.10.1.0 and is device type HA range extender 

Does "HA range extender" mean it only supports routing Zigbee HA (Home Automation) profile or does it also support routing all Zigbee 3.0 profiles?

MattWestb, 2021-08-24:
In Silabs SS5 is under HA devices (Lights is under ZLO = Zigbee Light and occupancy) and all is Zigbee 3 then using one not to old SDK.
All routers is routing all traffic for its PAN its joined to so long the devices is "normal Zigbee" ala standard only ZGP is need laving support for the extra proxy cluster.

One Zigbee router device (all standard from 1.0 and forward) dont care wot device type only that the frames its receiving is OK (network key and frame counter) then its forwarding the frame in direction to the receiver address.

Hedda, 2021-11-29:
_Originally posted by @xsp1989 in https://github.com/xsp1989/zigbeeFirmware/issues/18#issuecomment-978001961_

> **_Another piece of good news is that routing firmware will be released soon_**

@xsp1989 So you have decided to build and release a separate firmware image for [ITead Zigbee 3.0 USB Dongle](https://itead.cc/product/zigbee-3-0-usb-dongle/) that will come preconfigured a dedicated "Zigbee Router" device?

Will you be able to configure it so that either the "reset button" or "boot button"  can be used for pairing/joining during the commissioning process similar to any Zigbee device?

xsp1989, 2021-12-06:
This is the routing firmware：
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/RouterForDongle.gbl

Hedda, 2021-12-06:
Cool! Which exact EmberZNet version is it based on and is any of the buttons mapped as a pairing button to enable join?

Maybe perhaps detail the steps for putting into joining mode? And steps to reset, etc.

Question also posted here -> https://github.com/xsp1989/zigbeeFirmware/issues/16#issuecomment-981368117

_Originally posted by @xsp1989 in https://github.com/xsp1989/zigbeeFirmware/issues/18#issuecomment-978001961_

> **_Another piece of good news is that routing firmware will be released soon_**

@xsp1989 So you have decided to build and release a separate firmware image for [ITead Zigbee 3.0 USB Dongle](https://itead.cc/product/zigbee-3-0-usb-dongle/) that will come preconfigured a dedicated "Zigbee Router" device?

Will you be able to configure it so that either the "reset button" or "boot button"  can be used for pairing/joining during the commissioning process similar to any Zigbee device?

MattWestb, 2021-12-06:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle/RouterForDongle

Hedda, 2021-12-07:
So version EZSP 6.7.10? ...sorry, version used is still not 100% clear when the version is not part of the filename.

@xsp1989 Suggest/recommend rename "**_RouterForDongle.gbl_**" to "**_efr32mg21_zigbee_router_6.7.10.gbl_**" or something similar.

Version in filename is also important if ever release another version.

@xsp1989 Any change you could add patch or configuration info so others can build and compile their own router firmware?

digiblur, 2021-12-08:
No issues flashing the router firmware on the 3.0 iTead Dongle.  Worked just fine and paired up into Zigbee2MQTT and appeared in the map.  Looking forward to one for the Sonoff Zigbee Bridge as a router as well.

Hedda, 2021-12-08:
> Looking forward to one for the Sonoff Zigbee Bridge as a router as well.

FYI, xsp1989 posted replies with some more info about Zigbee Router firmware for ITead Sonoff ZBBridge SM-011 module here:

https://github.com/xsp1989/zigbeeFirmware/issues/2#issuecomment-988690014

and here:

https://github.com/arendst/Tasmota/discussions/11536#discussioncomment-1772584

Hedda, 2022-02-17:
FYI, @xsp1989 has uploaded/shared a signed Zigbee Router firmware for ITead's Sonoff ZBBridge to his google drive here:

> https://drive.google.com/file/d/1H-M5CTh_XGGVl2te4vHLQ5SfGPHL6yCS/view?usp=sharing
> 
> This is the signed routing firmware used by ZBB, the usage method is the same as the unsigned firmware.
> 
> https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md

PS: This idea also discussed specifically for ITead's Sonoff ZBBridge hacked with Tasmota here -> https://github.com/arendst/Tasmota/discussions/11536

Hedda, 2022-02-18:
Update from xsp1989 on router firmware posted on https://github.com/arendst/Tasmota/discussions/11536

"This firmware can ground the PA00 5S into pairing mode, and ground the RST to restart routing. UART is used to transmit data transparently.

and

"There is also another version of the routing firmware that can control the join network through a proprietary serial port protocol"

digiblur, 2022-02-18:
I was able to get it to work and join the network yesterday. That serial command one would be cool.

Hedda, 2022-02-18:
> I was able to get it to work and join the network yesterday.

@digiblur Nice! Can we look forward to a new blog article and YouTube video a'la https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html and https://www.digiblur.com/2021/11/tasmota-switch-control-zigbee-bulbs.html ? ;-)

Assume that you only tested joining to [Zigbee2MQTT](https://www.zigbee2mqtt.io/) and not [Home Assistant's ZHA](https://www.home-assistant.io/integrations/zha) and not [Zigbee2Tasmota (MQTT mode)](https://tasmota.github.io/docs/Zigbee/)?

How does it look as a Zigbee Router in Zigbee2MQTT, etc.?

PS: I'm guessing that users from other communities also using Zigbee, like OpenHAB and Domoticz, would also be interested :-D

digiblur, 2022-02-18:
Yes. Working on a guide. I did a converter file for Z2M and it was fine. I haven't tried it with ZHA yet. I will have to fire up my test system as I don't run it.

Hedda, 2022-02-20:
> Yes. Working on a guide. I did a converter file for Z2M and it was fine. I haven't tried it with ZHA yet. I will have to fire up my test system as I don't run it.

Nice! Check out this new "How to Convert the Sonoff Zigbee Bridge into a Router / Repeater" article by @digiblur here:

https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html

xsp1989, 2022-09-05:
Routing firmware has been released, so close

closed 2022-09-05 by xsp1989