xsp1989/zigbeeFirmware #18: EZSP 6.10.0.0 GA

MattWestb, 2021-06-19:
Silabs have released [Zigbee EmberZNet SDK 6.10.0.0 GA](https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.0.0.pdf). 

But i have not finding any interesting fixes in the Zigbee stack but in the Gecko Platform 3.2.0.0 GA (Simplicity Studio 5) i have finding some interesting fixes that perhaps can helping:

- 703231 Fixed an issue in Power Manager on Series 2 devices where we could get stuck waiting for HFXO to be ready when a hardware request on HFXO was removed in the middle of the restore process in Power Manager.
- 675253 Fixed an issue in Power Manager where, if the HFXO startup failed while actively waiting for it to be ready in a critical section, we could wait indefinitely
- 697483 Fixed bug affecting CTUNE for HFXO on Series 2 devices.

I think the 2 first is not relevant but the last looks "fitting" in the picture.

I think it can being worth one shut testing is its helping users that still having problems with there sleeping end devices.

Wot i can see in ZHA was the EZSP 6.7.8.0 helping many users with this problem and the CTune fix in 6.7.9.0 was also doing but its still some users that having fast battery draining (48 hours = OTA updating all the time for 2 days).

I dont have the device but have running ZHA with one tuya MG21 module that have the same chip and its working great with cooked EZSP 6.7.9 0 and 6.9.2.0 on one WeMos D1 Mini running tasmota or ESPHome for weeks with 7 different IKEA remotes.

Then Silabs have updating the compilers i think it can being good doing extra testing before relenting new firmware then its very likely getting new bugs implanted that can needing time for being pinpointed and fixes from Silabs side.

And one great thanks for supporting the community and i think in the end its one win win situation for both !!!

Mvh  Mattias

Hedda, 2021-08-24:
+1 requesting EmberZNet 6.10.0.0 (Silabs Zigbee NCP EmberZNet 6.10.0) firmware for ITead Zigbee 3.0 USB Dongle

Hedda, 2021-08-24:
Again, I also like to suggest looking at increasing default configuration parameters in a "pro" firmware as suggested in https://github.com/xsp1989/zigbeeFirmware/issues/6 and https://github.com/xsp1989/zigbeeFirmware/pull/13

Configuration Parameter | Value
-- | --
Part | EFR32MG21A020F768IM32
Version | EZSP 6.10.0.0
CTUNE value | 128
Address Table Size | 32
Child Table Size | 64
Source Routes | 200
TX | PB01
RX | PB00
CTS | ?
RTS | ?
BTL | PA00 (Low = BTL boot)
LED | PC00 

As I understand these are the "pro" configuration that tube0013 use in his EFR32MG21 Zigbee Coordinator Gateway devices (or?):

 # Specific Configuration for the Tube EFR32 Gateways
Because the EFR32 gateways uses some firrmware settings different than the bellows defaults it is recommended set them in the configuration.yaml so bellows will utilize them.

## For EFR32 Series 2 Gateways with Firmware based on the 6.9.2 SDK please use the following config
**Add these lines to your configuration.yaml file:**

```
zha:
  zigpy_config:
    source_routing: true
    ezsp_config:
      CONFIG_APS_UNICAST_MESSAGE_COUNT: 64
      CONFIG_MAX_END_DEVICE_CHILDREN: 32
      CONFIG_SOURCE_ROUTE_TABLE_SIZE: 200
      CONFIG_ROUTE_TABLE_SIZE: 16
      CONFIG_ADDRESS_TABLE_SIZE: 32
      CONFIG_PACKET_BUFFER_COUNT: 250
      CONFIG_BINDING_TABLE_SIZE: 32
      CONFIG_NEIGHBOR_TABLE_SIZE: 26
```

MattWestb, 2021-08-24:
`Child Table Size | 64` is out of question in all my firmware i shall being standard or being lowered.
It need being sett >0 for adding the first device (also routers) and then it can being sett to zero in ZHA config so all end devices is being blocked for using the coordinator as there parent and forcing them using routers then its 64 blocks or RAM allocated and not used = resource locked and not used that an being used for better things (if being configured in the firmware is allocated of  the CPU and if being used or not by the host system) like buffer for the Zigbee stack.

I is running my production and 2 test network with Child Table Size 1 for adding the first routers and then zero and its working great and the end device is not leaving the network if the coordinator is offline for days and coming back on line.

Forcing SED using routers and have good working routers (no old OSRAM Plugs and so on) in range is the key getting one Zigbee mesh working OK and with SM-011 modules its very importing then having problems with the stability of the module and WiFi interference from the ESP32.

And the some parameters is not helping at all then its out of range for standard Zigbee certified device and then the mesh is using the certified parameters that Raul was writing and then its miss using the resources on the chip.

Hedda, 2021-11-16:
@xsp1989 any plans for releasing newer Silabs EmberZNet NCP firmware that has been tested with Sonoff EFR32 adapter?

EmberZNet 6.10.3.0 GA should really be mature and stable enough now as only contain bug-fixes.

FYI, Zigbee EmberZNet 6.10.0.0 was just released on the 16th June of 2021 and newer EmberZNet 6.10 updates since then:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.0.0.pdf

EmberZNet 6.10.2.0 was released on the 8th of September of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.2.0.pdf

EmberZNet 6.10.1.0 was released on the 21st June of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.1.0.pdf

EmberZNet 6.10.3.0 was released on the 13th October of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.3.0.pdf

Hedda, 2021-11-16:
FYI, Elelabs has now released EmberZNet v6.10 firmware for their EFR32 based adapters (based on EFR32MG12 though):

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/blob/master/data/EFR32MG13/ELE_MG13_zb_ncp_115200_610_211112.gbl

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/31f8e9c09d612734edbe41619240162fe595ecc5

### Added

- 6.10 Zigbee Firmware for ELU013/ELR023 (just keep up with the latest EmberZnet SDK)
Their firmware update tool has also been updated with a few new features:

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/721019cb542b273159372357a30251330e7c8d51

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/e9b3fe1dd73b46fef1629a7a54ca0a257c1388ae

### Changed

- `ele_update` parameters changed. Now the function only updates to latest version of available Zigbee or Thread firmware
- `probe` and `restart` functions can now detect Thread adapters as well

xsp1989, 2021-11-24:
@Hedda @MattWestb EmberZNet v6.10 firmware it's OK
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/ncp-uart-sw_v6.10.3_115200.gbl

xsp1989, 2021-11-24:
Another piece of good news is that routing firmware will be released soon

MattWestb, 2021-11-24:
@s-hadinger interested testing if its working well on the ZBB that is SWD flashed ?

s-hadinger, 2021-11-24:
I don't have much time right now for Zigbee testing. I'm deeply involved in Berry scripting for Tasmota and some other projects

closed 2021-12-07 by xsp1989