xsp1989/zigbeeFirmware #20: [REQUEST] EmberZNet 7.0.0.0 (7.0.0.x) Zigbee Coordinator firmware images?

Hedda, 2022-01-11:
Silabs have now released Silicon Labs Zigbee EmberZNet SDK 7.0.0.0 (and matching Gecko SDK Shite 4.0.0 for it)

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.0.0.0.pdf

http://docs.silabs.com/zigbee/7.0

https://docs.silabs.com/zigbee/7.0/version-history

Gecko SDK Suite 4.0 release is also the first time Silicon Labs posted Gecko SDK Suite to a publicly available repository on GitHub:

https://github.com/SiliconLabs/gecko_sdk/tree/gsdk_4.0/protocol/zigbee

https://github.com/SiliconLabs/gecko_sdk/

https://github.com/SiliconLabs/gecko_sdk/releases

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/demo-applications.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/documentation.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/gecko-sdk.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/pintool.zip

https://github.com/SiliconLabs/gecko_sdk/archive/refs/tags/v4.0.0.zip

https://github.com/SiliconLabs/gecko_sdk/archive/refs/tags/v4.0.0.tar.gz

Update: Looks like it was released publicly on purpose.

https://community.silabs.com/s/article/Notes-on-the-recent-Simplicity-Studio-version-5-3-0-0-release?language=en_US

https://github.com/SiliconLabs/gecko_sdk/releases

Link to 4.0 ALSO seems to work through:

https://docs.silabs.com/gecko-platform/4.0/

Hedda, 2022-08-08:
FYI, grobasoz released Zigbee EmberZNet 7.0.1 firmware about 6-months ago and some report it working well for them:

https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32%20Series%202/EFR32MG21A020F1024

Some others are however reporting that Zigbee EmberZNet 6.10.3 is more stable for them in their existing environments.

Hedda, 2022-08-29:
> Is there a way to build and compile the firmware yourself/myself?

Yes but you first need to buy and register one of the official Silicon Labs compatible Zigbee development kit (dev kit) or starter kits in order to get the license need to use the EmberZNet Zigbee SDK, see https://www.silabs.com/development-tools/wireless where I think the least expensive is the EFR32xG22 for $99, see https://www.cnx-software.com/2020/06/25/efr32xg22-wireless-gecko-starter-kit-targets-bluetooth-zigbee-and-proprietary-wireless-development/?amp=1

Hedda, 2022-09-01:
@xsp1989 FYI, Zigbee EmberZNet SDK 7.0.2.0 GA (7.0.2) have now also been released as a bug-fix release:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.0.2.0.pdf

xsp1989, 2022-09-02:
@Hedda @MattWestb This link is an untested NCP firmware built on GSDK 4.1.1（EmberZnet 7.1.1.0） https://drive.google.com/file/d/1fIiGw9upxRoi7uoSs1zO2YRIAz0PG1UZ/view?usp=sharing

Calimerorulez, 2022-09-02:
Flashing the Sonoff Zigbee-E dongle went fine; will test if I have time.

Hedda, 2022-09-02:
Note that really want 7.0.x (7.0.2 or 7.0.1) release too as well because the newer 7.1.0 is not as backwards compatible as 7.0.x releases

xsp1989, 2022-09-04:
> > Is there a way to build and compile the firmware yourself/myself?
> 
> Yes but you first need to buy and register one of the official Silicon Labs compatible Zigbee development kit (dev kit) or starter kits to your (free) silabs account in order to get the license needed to access Zigbee EmberZNet SDK, see https://www.silabs.com/development-tools/wireless where I think the least expensive is the EFR32xG22 devkit for $99 (US-dollar), see https://www.cnx-software.com/2020/06/25/efr32xg22-wireless-gecko-starter-kit-targets-bluetooth-zigbee-and-proprietary-wireless-development/?amp=1
> 
> The Gecko SDK (GSDK) is otherwise available on Silicon Labs GitHub but again you still need to have registered a development board to show that you own one of their official products:
> 
> https://github.com/SiliconLabs/gecko_sdk
> 
> https://github.com/SiliconLabs

Perhaps the firmware can be built via the slc-cli command line and does not require registration。

https://www.silabs.com/documents/public/user-guides/ug520-software-project-generation-configuration-with-slc-cli.pdf

xsp1989, 2022-09-05:
> Note that really want 7.0.x (7.0.2 or 7.0.1) release too as well because the newer 7.1.0 is not as backwards compatible as 7.0.x releases

This Link is 7.0.2
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.0.2.0_115200.gbl

Hedda, 2022-09-07:
> @Hedda @MattWestb This link is an untested NCP firmware built on GSDK 4.1.1（EmberZnet 7.1.1.0） https://drive.google.com/file/d/1fIiGw9upxRoi7uoSs1zO2YRIAz0PG1UZ/view?usp=sharing

FYI, started a discussion about building and compiling "SLC-CLI" command line tool here -> https://github.com/zigpy/zigpy/discussions/1042

I guess it could be beneficial even if registration is needed with that too as could publish a batch file with batch script on GitHub for building and compiling "standard community firmware images" so commands could be shared within the open source home automation communities. 

That could be somewhat similar to how Koenkk shared patches and compilation instructions for Z-Stack ZNP firmware:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

faronov, 2022-09-09:
> Yes but you first need to buy and register one of the official Silicon Labs compatible Zigbee development kit (dev kit) or starter kits to your (free) silabs account in order to get the license [...]

As I know you can just add SDK in Simlicity Studio and it will be available without any DEVKIT registration

MattWestb, 2022-09-09:
GSDK is public on GitHub so testing installing Simplicity Studio 5 and installing the GSDK 7.x in it and look if you can cooking Zigbee firmware (examples in included).
If working OK = great and if not then sad but we is knowing.

I think one ZHA user have cooking somthing for one Thunder board II and its work working but i dont knowing if hi can doing all things in SS5.

I have one registered WSDK so i have licence for little more so cant testing it without deleting all Silabs things.

faronov, 2022-09-09:
> GSDK is public on GitHub so testing installing Simplicity Studio 5 and installing the GSDK 7.x in it and look if you can cooking Zigbee firmware (examples in included). If working OK = great and if [...]

same things, but I previously tested on some closed SDK when I didn't have such a license 
some SDK available on the git

ssypchenko, 2022-12-19:
Could you please compile 7.0.x version for OTA flash for ZB-GW03 device? The only way to flash it is with Tasmota, but flashing .gbl file leads to error :(

xsp1989, 2022-12-23:
If tasmota does not have new features that must be supported by 7.0.2, I may not release a new NCP firmware for ZB-GW03

MattWestb, 2022-12-23:
If i remember is EZSP 7.0.x using protocol version 8 and 7.1.X and higher is using version 9.
If host system (tasmota) is doing it OK it shall using V8 but i dont knowing if it the firmware is backward compatible and if tasmota is handling it OK.

ZHA (and i think also Z2M) have adding support for V9 so shall not being any problems.

Also if installing protocol version 9 firmware **can** braking tasmota firmware updating of the Zigbee module and need using other updating tool for getting back to working version.

Also EZSP 7.X is still very experimental and is not recommended for normal users and production systems.

Only 6.7.9.0 - 6.7.11 and 6.10.3.0 is recommended versions for being used in our community systems.

In the end if the manufacture and the community is **not** recommending one firmware, you is on your own then flashing other firmware and braking your device and dont have any support or grantee on the hardware.

ssypchenko, 2022-12-23:
Thank you for your comments, guys. The latest version working on ZB-GW01 I managed to find is 6.7.9. It seems to fix my issue when all devices go offline within a couple of hours from the start. However, I still have an issue when at one moment, all devices just disappeared from z2m. So I thought that version 7 would fix it. 
Anyway, thank you for your efforts to make our life smarter with these devices. :)

Hedda, 2022-12-24:
> If i remember is EZSP 7.0.x using protocol version 8 and 7.1.X and higher is using version 9. If host system (tasmota) is doing it OK it shall using V8 but i dont knowing if it the firmware is [...]

That is correct. Tasmota and also OpenHAB do not yet support EZSP v9 so can not use EmberZNet 7.1.x but can use EmberZNet 7.0.x since it still has EZSP v8

xsp1989, 2022-12-26:
> Thank you for your comments, guys. The latest version working on ZB-GW03 I managed to find is 6.7.9. It seems to fix my issue when all devices go offline within a couple of hours from the start. However, I still have an issue when at one moment, all devices just disappeared from z2m. So I thought that version 7 would fix it. Anyway, thank you for your efforts to make our life smarter with these devices. :)

Under normal circumstances, silicon Labs releases a new SDK, or adds a new function, which will introduce additional strange bugs (including simplicity studio, which will have strange problems every time it is updated), so the firmware we use for production has also been maintained In 6.7.10. In general, it is not a good suggestion to fix old problems with new versions

MattWestb, 2022-12-26:
I have the same experience as @xsp1989 with Silabs (G)SDKs and Simplicity studio (SS).
In the community is the latest bug fixed EZSP 6.7.X **is** the stable production version.
Aspirant is the latest 6.10.X but is not in the same class as 6.7.

SS5 and GSDK is Sirlabs redoing the code base for from the old `ember` to `SL` standard and and is braking many things and introducing many new bugs and all functions is `not` implanted in GSDK7.X.
One example making one NCP for My Billy project and cant getting it using the latest tool change then its installed and active and SS5 is using 10.21. After 2 days i deleting the project and making one new work space and setting up one new project with the same hard and software setting and it using tool change 10.31 = 2 days wasted and not productive.

So GSDK 7.X is best in Beta stage and in many parts Alpha and not ready for production systems.

For SonOff as company is make no sense cooking firmware that is not working and getting problem with support for there customer then the system is not working OK.

I think better only releasing stable and stable candidate NCP firmware.
As one project for future releasing Alpha / test firmware for RCP / multi protocol 801.15.4 can being good but still is not for production and its no warranty on it until flagged production quality from Silabs and company testing.

Im very grateful that SoOff is supporting the coordinator products with firmware updates for fixing problems and also adding futures that most companies is not doing and also helping the community getting our system working better.

xsp1989, 2023-01-03:
@MattWestb I'm working for easyiot now, my views don't represent those of sonoff, thanks.

MattWestb, 2023-01-03:
Great so more Raspberry PI and Arduino :-))

Thanks for the help you have supporting the community with and great time at the new company @xsp1989 !!

Hedda, 2023-01-23:
FYI, Silicon Labs released Gecko SDK (GSDK) Version 4.1.4.0 as a bug-fix release that contains Zigbee EmberZNet SDK 7.1.4.0:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.4

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.4.0.pdf

xsp1989, 2023-04-27:
The 7.1.4.0 firmware has been re-released now, based on GSDK4.1.4

closed 2023-08-24 by xsp1989