xsp1989/zigbeeFirmware #22: Firmware compatibility with ITead's new  “ZBDongle-E” Zigbee 3.0 USB Dongle Plus adapter based on the EFR32MG21?

Hedda, 2022-08-07:
@xsp1989  Can you tell if existing firmware releases are compatible with ITead's new “ZBDongle-E” that is also based on EFR32MG21?

https://itead.cc/product/zigbee-3-0-usb-dongle/

I guess my question is if it this new “ZBDongle-E” dongle board is pin-compatible with the first/old ITead bareboe Zigbee 3.0 USB Dongle (model "9888010100045")  as well as on CoolKit Technologies “SM-011 V1.0” module used in the ITead Sonoff ZBBridge Zigbee Bridge?

Here is by the way link to FCC filing/report for the new “ZBDongle-E” (a.k.a. "ZBD-E" ) adapter which contains copy of the user manual + a lot more detailed + many more internal and external pictures:

[https://fccid.io/2APN5ZBD-E](https://fccid.io/2APN5ZBD-E)

[https://fccid.io/2APN5ZBD-E/User-Manual/User-Manual-6012461.pdf](https://fccid.io/2APN5ZBD-E/User-Manual/User-Manual-6012461.pdf)

[https://fccid.io/2APN5ZBD-E/Internal-Photos/Internal-Photos-6012455](https://fccid.io/2APN5ZBD-E/Internal-Photos/Internal-Photos-6012455)

[https://fccid.io/2APN5ZBD-E/External-Photos/External-Photos-6012451](https://fccid.io/2APN5ZBD-E/External-Photos/External-Photos-6012451)

It at least looks to use the same [EFR32MG21A020F768IM32 (EFR32 Mighty Gecko Series 2) SoC](https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs/device.efr32mg21a020f768im32?tab=specs) used on the first ITead Zigbee 3.0 USB Dongle, as well as on CoolKit Technologies "SM-011 V1.0" module used in th ITead Sonoff ZBBridge Zigbee Bridge and in Tuya ZS3L and Tuya ZSLC5 SoC modules. It includes a 80 MHz ARM Cortex-M33 core, 768 Flash Storage, 64 RAM, in a QFN32 package, which radio has + 20 dBm maximum power output and receive sensitivity of -104 (250 kbps O-QPSK DSSS) dBm. 

There are further details and more users discussion here:

https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695

and here:

https://github.com/Koenkk/zigbee2mqtt/discussions/13373

[image]

By the way, news is that a similar Silabs EFR32MG21 SoC will also be used in the official "Home Assistant SkyConnect USB Stick", see:

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594

You can submit your interest in the official "Home Assistant SkyConnect USB Stick" here if want to get mailed updates on that dongle:

https://docs.google.com/forms/d/e/1FAIpQLScEjHSBJszUZfgO3MIDO51IHr3Oeohcs8BLpRIjY1liJ58IpA/viewform

Hedda, 2022-08-09:
I asked ITead and was told that ZBDongle-E ships with Zigbee NCP 6.10.3 firmware that has enabled 20 dBm output RF power default, and is not hardcoded so users/applications can choose to lower output RF power.

FYI, got also this from ITead:

Configuration Parameter values Table for Silicon Labs EmberZNet PRO Zigbee networking protocol stack:

[image]

Hedda, 2022-08-15:
New firmware as linked to by ITead:

https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E

https://sonoff.tech/product-review/sonoff-zigbee-3-0-usb-dongle-plus-tutorials/

https://sonoff.tech/wp-content/uploads/2022/08/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf

xsp1989, 2022-09-01:
hi @Hedda ZBDongle-E is developed by other teams of Sonoff, not sure if the firmware is compatible, and there are no plans to release firmware for it.

closed 2022-09-05 by xsp1989