xsp1989/zigbeeFirmware #24: clarification on the EZSP configuration parameter

pipiche38, 2022-08-09:
In case it is the wrong place to ask this question feel free to remove it.

I'm trying to understand this concept of all of the EZSP config parameter, and I also understood that we can use a yaml file to change that configuration

In your latest firmware update, you have increase those following parameter inside the firmware.

Why not doing with yaml config file ?
What is the Yaml config file override it ? with a lower or bigger value ?
Is there any relationship between the table , like if we increase one and other one should also be increase ?

Parameter | size
--|--
Address Table Size | 32
Child Table Size | 32
Source Routes | 200

MattWestb, 2022-08-09:
All ERF32 SOC (its one ARM CPU) is allocating the memory then the firmware is compiled and if not doing anything its using this parameters.

If setting "software" (command parameters) then the SOC is using them **if** its decreasing the memory used.

Example the firmware is having 50 Source Routes and trying allocating 200 you is getting one error then the memory map is not having place for the 150 extra.

If the firmware is having 200 and setting 50 its being happy and using 50 and have the rest of memory unused and cant being used for other things.

pipiche38, 2022-08-09:
Understood. 

here is the current config file. If I well understood I need to put `source_routing to True` in the config file, but I do not set CONFIG_SOURCE_ROUTE_TABLE_SIZE to a specific value, the value of 200 (decimal) will be used.
Is that correct ? Or as CONFIG_SOURCE_ROUTE_TABLE_SIZE is defined to 200 in the firmware, I even don't have to do anything with the `source_routing` parameter

```

bellows -d /dev/ttyAMA0 --baudrate 115200 config --all
NOTE: Configuration changes do not persist across resets
CONFIG_PACKET_BUFFER_COUNT=250
CONFIG_NEIGHBOR_TABLE_SIZE=16
CONFIG_APS_UNICAST_MESSAGE_COUNT=10
CONFIG_BINDING_TABLE_SIZE=32
CONFIG_ADDRESS_TABLE_SIZE=16
CONFIG_MULTICAST_TABLE_SIZE=8
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
CONFIG_SOURCE_ROUTE_TABLE_SIZE=200
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_END_DEVICE_BIND_TIMEOUT=60
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_CERTIFICATE_TABLE_SIZE=0
CONFIG_APPLICATION_ZDO_FLAGS=0
CONFIG_BROADCAST_TABLE_SIZE=15
CONFIG_MAC_FILTER_TABLE_SIZE=0
CONFIG_SUPPORTED_NETWORKS=1
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=1
CONFIG_ZLL_RSSI_THRESHOLD=128
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=9
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=360
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
```

MattWestb, 2022-08-10:
Its working in ZHA with only setting the source_routing to true with bellows and the NCP is  starting doing routing discovery broadcasts to all routers.

I can recommending setting `CONFIG_MAX_END_DEVICE_CHILDREN=0` **after** adding the first router device and the NCP cant having direct children and you is getting less problems with leaving sleepers and battery draining of remotes.

pipiche38, 2022-08-10:
Very interesting, and that is great to be able live to change the config setting .  By that you are forcing End device to get connected to routers and not directly to the Coordinator.

closed 2022-08-10 by pipiche38

MattWestb, 2022-08-10:
Yep IKEA controllers is having one bug (Sirlabs have fixing it but IKEA is having not updated the stack enough) and also Philips HUE remotes can not going in sleep mode and is burning the battery in some days if they is jumping to one new router.

All coordinator is also one router but the TC and other things is making them not good for sleepers and if the coordinator is off line (restart or system down) Xiaomi /Aqara sleepers is leaving the network if they is trying communicating with the network and its parent is off line or not communicating OK.