xsp1989/zigbeeFirmware #25: Bluetooth controller firmware for EFR32MG21 compatible with Home Assistant's new native Bluetooth integration?

Hedda, 2022-09-07:
Could a Bluetooth firmware be built for EFR32MG21 that is compatible with Home Assistant's native Bluetooth integration?

https://www.home-assistant.io/integrations/bluetooth/

That is, could you build a Bluetooth controller firmware with Bluetooth Low Energy (BLE) support for EFR32MG21 adapters?

Home Assistant 2022.8 August 2022 release now got first-class Bluetooth support (though still not got many integrations using it):

https://www.home-assistant.io/blog/2022/08/03/release-20228/#first-class-bluetooth-support

Read EFR32MG2x multi-protocol SoCs support Bluetooth but have not seen any Bluetooth adapter for computers based on them?

https://www.silabs.com/wireless/bluetooth

Would be awesome if could flash existing 20 dBm EFR32MG21 adapters sold for Zigbee with "Bluetooth Long Range" firmware.

Home Assistant currently lists known working Bluetooth adapters based on BCM20702A1, RTL8761BU/RTL8761B, CSR8510A10:

https://github.com/home-assistant/home-assistant.io/blob/next/source/_integrations/bluetooth.markdown#known-working-adapters

PS: Home Assistant 2022.9 (September 2022) release will also add support for using multiple Bluetooth adapters at the same time:

https://github.com/home-assistant/home-assistant.io/blob/next/source/_integrations/bluetooth.markdown#multiple-adapters

Hedda, 2022-09-16:
@xsp1989 Please see this discussion if you have time and might be interested in this here -> https://github.com/grobasoz/zigbee-firmware/issues/31

Again, my question is regarding the possibility of optionally flashing an EFR32MG21 based adapter (like easyiot model ZB-GW04) with only a Bluetooth "controller" firmware in order to permanently make it into a dedicated Bluetooth adapter only. 

PS: I was not thinking about using it as a multiprotocol adapter with RPC firmware to use Zigbee and Thread at the same time.

xsp1989, 2022-09-17:
@Hedda Thanks for your support, I built a HCI firmware for GW04, but I can't add it to homeassistant. If you have any other news, please let me know and I will continue to research.

closed 2023-01-03 by xsp1989