xsp1989/zigbeeFirmware #29: ZBdongle-E

XTimt, 2022-10-27:
Hello.
I have a problem updating the ZBDongle-E firmware, please help.

root@ubuntu-xt:~/elelabs-zigbee-ezsp-utility# python3 [Elelabs_EzspFwUtility.py](https://vk.com/away.php?to=http%3A%2F%2FElelabs_EzspFwUtility.py&cc_key=) flash -f data/ncp-uart-sw_7.1.1.0_115200.gbl -p /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220718134126-if00
2022/10/27 09:59:59 Elelabs_EzspFwUtility: Generic Zigbee EZSP adapter detected:
2022/10/27 09:59:59 Elelabs_EzspFwUtility: Firmware: 6.10.3-41
2022/10/27 09:59:59 Elelabs_EzspFwUtility: EZSP v8
2022/10/27 09:59:59 Elelabs_EzspFwUtility: Launch in bootloader mode
2022/10/27 10:00:09 Elelabs_EzspFwUtility: EZSP adapter in bootloader mode detected:
2022/10/27 10:00:09 Elelabs_EzspFwUtility: Gecko Bootloader v1.12.00
2022/10/27 10:00:10 Elelabs_EzspFwUtility: Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
send error: expected ACK; got b'\x18' for block 1
send error: expected ACK; got b'\x18' for block 1
send error: expected ACK; got b'\x18' for block 1
send error: expected ACK; got b'\r' for block 1
send error: expected ACK; got b'\n' for block 1
send error: expected ACK; got b'S' for block 1
send error: expected ACK; got b'e' for block 1
send error: expected ACK; got b'r' for block 1
send error: expected ACK; got b'i' for block 1
send error: expected ACK; got b'a' for block 1
send error: expected ACK; got b'l' for block 1
send error: expected ACK; got b' ' for block 1
send error: expected ACK; got b'u' for block 1
send error: expected ACK; got b'p' for block 1
send error: expected ACK; got b'l' for block 1
send error: expected ACK; got b'o' for block 1
send error: expected ACK; got b'a' for block 1
send error: NAK received 17 times, aborting.

2022/10/27 10:00:17 Elelabs_EzspFwUtility: Firmware upload failed. Please try a correct firmware image or restart in normal mode.

xsp1989, 2022-11-01:
@XTimt Dongle-E is not maintained by our team, you can go here for help：
https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/issues

TheJulianJES, 2022-11-05:
I know it's not officially supported but I've had no issues running this firmware on it: https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

MattWestb, 2022-11-06:
Witch version of the firmware is working OK latest 7.1.1.0 ?
Im curious :-)

TheJulianJES, 2022-11-06:
Now on 6.10.3 but 7.0 and 7.1 also worked fine mostly. There seems to be a bug where SUPPORTED_NETWORKS=2 cannot be "downgraded" to 1 by bellows for 7.x, so the option has to be set to 2 for ZHA to start (same as what the firmware was compiled with). Z2M does not touch this

MattWestb, 2022-11-06:
You can making one PR for Bellows taking the default away then init the NCP then the NCP is using the firmware default and its being easier for not so technical users or we must looking then user have problem and helping them.

I have not cooking 7.X NCP firmware for Billy then its not sported and the new MG21 (Marcus) is not easy getting working then the hardware is locked and i trying getting the RCP working OK for all my coordinators.

For my 6.10.3.0 looks working stable for all MG1 and 2 devices so its one stable candidate and its good getting feed back from others if we shall upgrading it to recommended in Zigps wiki.

Thanks for feedback and also for helping the community users.

Hedda, 2022-11-22:
7.1.1.0 has also been reported to work with Zigbee2MQTT and IoBroker via zigbee-herdsman -> https://github.com/Koenkk/zigbee-herdsman/issues/319

> I know it's not officially supported but I've had no issues running this firmware on it

FYI, know that grobasoz also builds unofficial community firmware builds for various of Silicon Labs based Zigbee adapters:

https://github.com/grobasoz/zigbee-firmware/

There is a discussion specifically about ITead’s “Sonoff Zigbee 3.0 USB Dongle Plus V2” (model “ZBDongle-E”, a.k.a. "Dongle-E"): 

https://github.com/grobasoz/zigbee-firmware/issues/28

Obviously use at your own risk applies to any such unofficial community firmware builds for any Zigbee adapter/dongle.

Hedda, 2022-11-22:
Another tip is to use alternative flashing tools for flashing firmware Silicon Labs EFR32 based adapters:

- [Elelabs Firmware Update Utility](https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/) (multi platform Python based command line tool)
- [puddly Universal Silicon Labs Flasher](https://github.com/puddly/universal-silabs-flasher) (multi platform Python based command line tool)
- [agners silabs-flasher - Silicon Labs Firmware flashing utility](https://github.com/agners/silabs-flasher) (multi platform Python based command line tool)
- [walthowd husbzb-firmware script](https://github.com/walthowd/husbzb-firmware) (community maintained multi platform bash script)
- [Manual Xmodem sending commands over a terminal console](https://sonoff.tech/wp-content/uploads/2022/08/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf) (note that any terminal application with "Xmodem(N)" send can be used)
- Silicon Labs [Simplicity Studio](https://www.silabs.com/developers/simplicity-studio) included "Flash Programmer" ([instructions](https://docs.silabs.com/simplicity-studio-5-users-guide/latest/ss-5-users-guide-building-and-flashing/flashing#flash-programmer)) (can't find your device? read below!)
- [Additional programming options for Silicon Labs MCU based devices](https://www.silabs.com/developers/mcu-programming-options)

Hedda, 2022-12-21:
> I know it's not officially supported but I've had no issues running this firmware on it: https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

FYI, when their new “Sonoff Zigbee 3.0 USB Dongle Plus V2” (model “ZBDongle-E”) was released I asked @Daniel-zhan-itead (product manager at ITead) about firmware compatibility with their old 9888010100045 (previous barebone Silabs EFR32MG21 based Zigbee 3.0 USB adapter from ITead) and got the reply that the pinout definition for the EFR32MG21 SoC chip on the new model ZBDongle-E board is exactly the same as the original first-generation barebone Zigbee 3.0 USB Dongle model 9888010100045, but that that the new model ZBDongle-E adapter can still not use the exact same firmware images as their old barebone model 9888010100045 adapter, so I guess that must mean that they might not use the exact same EFR32MG21 SoC chip model or?

Back when this new "ZBDongle-E" was first released I asked ITead about firmware compatibility and got the reply that the pinout definition for the EFR32MG21 SoC chip on the new model ZBDongle-E board is exactly the same as the original first-generation barebone Zigbee 3.0 USB Dongle model 9888010100045, but they answered that the new model ZBDongle-E adapter can still not use the exact same firmware images as their old barebone model 9888010100045 adapter, so I guess that must mean that they do not use the exact same EFR32MG21 SoC chip model or?

As far as I can tell the ZBDongle-E does use the exact same [EFR32MG21A020F768IM32 SoC ](https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs/device.efr32mg21a020f768im32?tab=specs) used on their previous model 9888010100045 ITead Zigbee 3.0 USB Dongle, 

EFR32MG21A020F768IM32 is well as on CoolKit Technologies “SM-011 V1.0” module used in the ITead Sonoff ZBBridge Zigbee Bridge and in Tuya ZS3L and Tuya ZSLC5 SoC modules. 

That includes an 80 MHz ARM Cortex-M33 core, 768 kB Flash Storage, 64 kB RAM in a QFN32 package

At that time I also asked ITead about the default RF output power configuration and was told that the new ZBDongle-E adapter currently ships with Zigbee NCP 6.10.3 firmware that has enabled 20 dBm output RF power default, however, is not hardcoded in the firmware so users/applications can choose to lower output RF power if they want.

In addition, got part of the Configuration Parameter values Table for Silicon Labs EmberZNet PRO Zigbee networking protocol stack:

|Configuration Parameter|Value|
| --- | --- |
|EMBER_APS_UNICAST_MESSAGE_COUNT|32|
|EMBER_PACKET_BUFFER_COUNT|250|
|EMBER_NEIGHBOR_TABLE_SIZE|26|
|EMBER_SOURCE_ROUTE_TABLE_SIZE|200|
|EMBER_ADDRESS_TABLE_SIZE|32|

It does not mention “Child Table Size” but I guess that is probably configured to 32 same as for old model 9888010100045 as per https://github.com/xsp1989/zigbeeFirmware/blob/4533c8daa1e571801a2c49dca8498a1b1b79f54f/firmware/Zigbee3.0_Dongle/EZSP/README.md or?

xsp1989, 2023-01-03:
dongle-E is not developed by us, this issue is not discussed here, so it is closed

closed 2023-01-03 by xsp1989