xsp1989/zigbeeFirmware #30: Sonoff Zigbee Bridge As Router - Maximum 6 devices can be paired

ZamliyMedre, 2022-11-10:
It seems we can only pair maximum of 6 devices with the Sonoff Zigbee Bridge flashed with the router firmware "efr32mg21_zigbee_router_signed-6.7.10.gbl.ota".

I currently have 1 Sonoff Zigbee Bridge (coordinator) and 2 Sonoff Zigbee Bridges flashed with router firmware mentioned above. Both routers successfully paired with the coordinator, but it seems I can only add maximum of 6 devices per router.

I have tried both Zigbee2Mqtt and Home Assistant ZHA, same results. After pairing the 6th device, it is not possible to pair any more with the target router. Nothing created in the logs for both Zigbee2Mqtt or HASSIO ZHA Integration during pairing it just times out for both.

Same hardware running with the Tasmota bridge firmware can support more than 6 child devices, for that not sure why we can only add maximum of 6 devices when running with the router firmware.

Any help to resolve the issue would be very much appreciated.

Many Thanks

Erdem

xsp1989, 2022-11-18:
Thank you very much for your feedback, the firmware on google driver may be the firmware compiled earlier, it should be the default 6 dormant devices, and now the firmware that supports 32 sub-devices has been updated：

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed/Router-Singed

ZamliyMedre, 2022-11-18:
Hi

Thank you for the update however I am having difficulty apply the updated firmware, Tasmota reporting invalid file signature, what am I doing wrong ? any help would be very much appreciated.

[image]
<img width="475" alt="Screenshot 2022-11-18 at 10 26 32" src="https://user-images.githubusercontent.com/64253
[image]
821/202681727-af77c61d-1220-42a8-bc72-697c04cae992.png">

ZamliyMedre, 2022-11-18:
Thank you very much @xsp1989 for the updated firmware!

I have tried to update the firmware on the existing Tasmota Sonoff Zigbee Router but this failed with "Invalid File Signature" error.

After some investigation it turns out you have to first flash the router with the tasmota-zbridge firmware > change the module type to "Sonoff ZBbridge (75)" after you can update the new firmware.

For visibility I have captured some screenshots for anyone had the same issue as me as a step-by-step guide to update already flashed Tasmota Sonoff Zigbee Router.

[image]

Change OTA url to : http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz

Click "Start Upgrade"

[image]

[image]

After device reboots, Click "Configuration"

[image]

Click "Configure Module" option

[image]

Select "Sonoff ZBbridge (75)" from drop down list, and click "Save"

[image]

[image]

After device reboots, Click "Firmware Upgrade"

[image]

Under "Update firmware by File Section", Select "Choose File" option to browse to the new firmware file downloaded.
Click "Start Upgrade"

[image]

[image]

[image]

After device reboots, Click "Consoles"

[image]

Click "Console"

[image]

Paste the following command to the text box and hit "Enter / Return Key"

backlog template {"NAME":"Sonoff ZigRouter","GPIO":[320,5088,0,5120,257,256,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":18} ; delay1 ; module 0

[image]

Wait for the device to restart, Click "Consoles" to go back to previous menu

[image]

Click "Main Menu" to go back to previous menu

[image]

Finally reboot the device again to complete upgrade.

[image]

ZamliyMedre, 2022-11-18:
@xsp1989 thanks again for updating the firmware, I have now successfully paired all Aqara Zigbee devices with Sonoff Zigbee Routers.

If anyone having issues with dropped Aqara door / window sensors especially the older model 11LM with temperature sensors built-in .. here is what I have done to keep them stick to Zigbee Router.

1. Get Sonoff USB Zigbee dongle plug that directly to the USB port of your device where HASSIO is running
2. HASSIO will detect it and ask you to install ZHA integration...
3. Start pairing new Sonoff Zigbee Routers, by pairing first one closest to the USB Coordinator.
4. than pair any other Sonoff Zigbee Routers by using the previous router you have paired so you basically daisy chain the Sonoff Routers...by distance.
5. If you have Aqara door sensors, these thing don't have great range for that you need to me maximum 7-10m away from the Zigbee Router, but first pair them right next to the Sonoff Router than you can put them in their permanent place. Make sure you select the relevant router closer to the sensor your pairing and enabling pairing on that router. This is very important as Aqara sensors do not like to roam and they stick to the initial coordinator / router they are paired to.
6. Gradually pair the rest of Aqara sensors according to the distance they are to the routers you may have.
7. Finally add any other Zigbee devices, again by using closest router to that device.

I hope this helps to other who had bad luck with the Aqara sensors.

Here is a screen shot of my Zigbee network with 43 devices, where I have 21 Aqara door /window sensors (11LM).

[image]

closed 2022-11-18 by ZamliyMedre

MattWestb, 2022-11-19:
@ZamliyMedre If you like after adding your backbone of rosters you can blocking the coordinator having direct children (end device cant using it and must using other routers) by adding this line in HA config under ZHA `      CONFIG_MAX_END_DEVICE_CHILDREN: 0` and restarting HA or the integration but you mus making it larger then zero if you is restarting from scratch for adding the first router or it cant joining.

Great work done !!