xsp1989/zigbeeFirmware #31: Home Assistant wont start with 7.X.X firmware (ERROR_OUT_MEMORY, and RESET_ASSERT)

xatr0z, 2022-11-27:
I'm not entirely sure if I have to report this here, at zigpy or home assistant, so sorry in advance if this is the wrong place. :)

I tried upgrading my Itead Zigbee stick ([model without antenna/case](https://zigbee.blakadder.com/ITEAD_9888010100045.html)) with the unsigned images.
**ncp-uart-sw_v6.10.3_115200.gbl** worked fine in HA, but didnt solve the bugs with aqara devices i am having. Therefore i tried **ncp-uart-sw_7.0.2.0_115200.gbl** and later **ncp-uart-sw_7.1.1.0_115200.gbl** . Flashing both went OK and device went back to normal mode. But I wasnt able to get it working in HA:

* `[bellows.ezsp.protocol] Couldn't set EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE=16 configuration value: EzspStatus.ERROR_OUT_OF_MEMORY`
  * fixed that by forcing it to 12 (setting it to 16 had no effect) with: `CONFIG_ADDRESS_TABLE_SIZE:              12`
* after that the next error appears: `Couldn't set EzspConfigId.CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=2 configuration value: EzspStatus.ERROR_OUT_OF_MEMORY`
  * tried setting to 2 and 1, but would only start with 0: `CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE:    0`
* Kept hitting different issues after that. So I tried different configs posted online:
  * From [issue at itead](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/issues/10)
    ```
      CONFIG_APS_UNICAST_MESSAGE_COUNT: 32
      CONFIG_PACKET_BUFFER_COUNT:       250
      CONFIG_NEIGHBOR_TABLE_SIZE:       26
      CONFIG_SOURCE_ROUTE_TABLE_SIZE:   200
      CONFIG_ADDRESS_TABLE_SIZE:        14
    ```
  * Different variations ie
    ```
      CONFIG_APS_UNICAST_MESSAGE_COUNT: 32
      CONFIG_PACKET_BUFFER_COUNT:       250
      CONFIG_NEIGHBOR_TABLE_SIZE:       26
      CONFIG_SOURCE_ROUTE_TABLE_SIZE:   16
      CONFIG_ADDRESS_TABLE_SIZE:        12
      CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE: 0
      CONFIG_MULTICAST_TABLE_SIZE:      12
      ```
  * As [advised by puddly](https://github.com/home-assistant/core/issues/82004#issuecomment-1312855293):
    ``` 
     CONFIG_ADDRESS_TABLE_SIZE:              16  # FW:  32, ZHA:  16
     CONFIG_MULTICAST_TABLE_SIZE:             8  # FW:   8, ZHA:  16
     CONFIG_PACKET_BUFFER_COUNT:            250  # FW: 250, ZHA: 255
     CONFIG_SOURCE_ROUTE_TABLE_SIZE:         16  # FW: 200, ZHA:  16
     CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE:  2  # FW:   0, ZHA:   2
     ```
  * Based on values in [the readme here](https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP):
    ```
      CONFIG_CTUNE_VALUE:   128
      CONFIG_ADDRESS_TABLE_SIZE:  32
      ```

...but none worked. I ended up with an unresponsive coordinator (RESET_ASSERT which i assume is some sort of crash):
```
2022-11-27 14:45:06.349 DEBUG (bellows.thread_0) [bellows.uart] Error code: RESET_ASSERT, Version: 2, frame: b'c2020682af7e'
2022-11-27 14:45:06.350 INFO (MainThread) [bellows.ezsp] NCP entered failed state. No application handler registered, ignoring...
2022-11-27 14:45:10.230 DEBUG (bellows.thread_0) [bellows.uart] Closed serial connection
```

Full error logs: [bellows_zigpy_logs___after_ezsp_config.txt](https://github.com/xsp1989/zigbeeFirmware/files/10099065/bellows_zigpy_logs___after_ezsp_config.txt)

This were the bellows defaults: [bellows_defaults.txt](https://github.com/xsp1989/zigbeeFirmware/files/10099060/bellows_defaults.txt)

PS: 
Eventually I tried to go back to 6.10.3 and encountered the same issue as #28 . The outlined steps (using onboard buttons) helped to be able to revert the firmware to

xsp1989, 2022-12-20:
This problem may be related to NVM cache, you can try to use nvm3_initfile.gbl file to reinitialize NVM.

MattWestb, 2022-12-20:
May i requesting (not for my but for our users in the future) then making new ZESP firmware patching the 2 above configuration parameters (or using more standard settings the your chip have enough ram and CPU) so not getting problems with the coordinators.

One link for more information that can being good then some parameters cant being set in the GUI in SS5 https://github.com/NabuCasa/silabs-firmware/issues/1#issuecomment-1016230203.

xatr0z, 2022-12-20:
> This problem may be related to NVM cache, you can try to use nvm3_initfile.gbl file to reinitialize NVM.

Thanks for the reply, I already tried the nvm3_initfile but it didnt help

TheJulianJES, 2023-01-03:
Set ``CONFIG_SUPPORTED_NETWORKS`` to ``2`` for the 7.x images from this repo.
Bellows wants to set this to 1 (lowering limits should always work). However, the firmware simply crashes.

closed 2023-04-26 by xsp1989