xsp1989/zigbeeFirmware #32: [REQUEST] Silicon Labs Concurrent Multiprotocol Zigbee in NCP mode and Open-Thread in RCP mode?

Hedda, 2022-12-12:
FYI, looks like Nabu Casa developers are working on NCP + RPC combination "beta" firmware for Home Assistant SkyConnect:

https://github.com/NabuCasa/silabs-firmware

https://github.com/NabuCasa/silabs-firmware/tree/main/EmberZNetAndOpenThreadRCP/beta

The reason for also wanting OpenThread RPC support is that Home Assistant 2022.12 has now been released with initial multiprotocol support Zigbee and Thread (including Matter over Thread), as tested with "Home Assistant SkyConnect" USB adapter and "Home Assistant Yellow" embedded radio:

https://www.crowdsupply.com/nabu-casa/home-assistant-yellow/updates/more-yellows-shipping-and-thread-plus-matter-updates

https://www.home-assistant.io/blog/2022/12/07/release-202212/

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594/

Silicon Labs Gecko SDK 4.1.0 was released with "_Alpha Concurrent Multiprotocol Zigbee in NCP mode and OpenThread in RCP mode_":

https://github.com/SiliconLabs/gecko_sdk/releases

I guess that mean could now build a unified firmware image that can support Zigbee in NCP mode and OpenThread in RCP mode?

If this is the case then such backwards-compatible firmware images might be interesting for users of existing applications that use "old school" Zigbee in NCP mode as it would open up the ability to concurrently also use the same radio adapter for OpenThread/Thread RPC (and thus also the upcoming Matter connectivity protocol), or does it not?

Please see a somewhat related discussion here (though it only discusses RPC mode and not NCP + RPC) -> https://github.com/zigpy/zigpy/discussions/894

Silabs Zigbee EmberZNet SDK 7.1.0.0 

Please take note that because of its new (Alpha) support for concurrently running firmware with Zigbee NCP and OpenThread RPC at the same time announced in Silicon Labs Gecko SDK v4.1.0 I do believe that sooner or later going with EmberZNet 7.1.x firmware or even in the relatively near future for early adopters will be tempting to both manufacturers of Zigbee radio USB adapters and end-users owning existing Silicon Labs EFR32MG12, EFR32MG13, and EFR32MG21 adapters.

As such firmware will mean that could use the same EFR32MGxx adapter as a Zigbee Coordinator for existing Zigbee devices and an OpenThread Border Router (OTBR) for the upcoming Matter over Thread (formerly Project CHIP / Connected Home over IP). So in essence users could just flash their existing Silabs Zigbee Coordinator and have it also act as a radio for Matter over Thread.

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.0

<html>
<body>

Component | Version | Summary | Details
-- | -- | -- | --
[OpenThread SDK](https://www.silabs.com/wireless/thread#software) | 2.1.0.0 | - Alpha SPI support for OpenThread RCP without CPC<br> - Thread 1.2 and 1.3 support for OpenThread<br> - Updated GCC compiler version to 10.3.1<br> - **Alpha Concurrent Multiprotocol Zigbee in NCP mode and OpenThread in RCP mode**<br> - Alpha Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode | [Release Notes](https://www.silabs.com/documents/public/release-notes/open-thread-release-notes-2.1.0.0.pdf)
[Zigbee EmberZNet SDK](https://www.silabs.com/wireless/zigbee#software) | 7.1.0.0 | - 2.4GHz Zigbee Smart Energy support for xG24<br> - 802.15.4 Signal Identifier and MAC CCA Mode 2 and 3 support for xG24<br> - Zigbee Green Power Gateway Backup<br> - Updated GCC and IAR compiler version<br> - **Alpha Concurrent Multiprotocol Zigbee in NCP mode and Open-Thread in RCP mode**<br> - Alpha Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode | [Release Notes](https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf)

</body>
</html>

PS: Note that importantly this EmberZNet SDK 7.1.0 added an update to EZSP v9 (EmberZNet Serial Protocol version 9) which might break existing Zigbee implementation if user Silabs Zigbee EmberZNet SDK 7.1.0.0 NCP firmware before updated to EZSP v9.

xsp1989, 2022-12-19:
Thank you for your suggestion. We also plan to support RCP, but we have not found a good test method and test equipment, so there has been no further development.

MattWestb, 2022-12-19:
@xsp1989 if only doing RCP for Silabs EFR32MGx is HA implanted Silabs reference design in https://github.com/home-assistant/addons/tree/master/silabs-multiprotocol and discussion is https://github.com/zigpy/zigpy/discussions/894.
That implanting the RCP for OTBR and EZSP with all NCP functionality on the host system ans is using the SOC as one radio with one multiplexer for different radio protocols so OTBR having one hadle and Zigbee is having one other and the add-on is then running OTRB (Open Thread Broader Router) for Thread aand one Zigbeed for making one normal EZSP / Zigbee to one network port that ZHA and other system can suing.
On bad thing with Silabs implementation is they is using one new protocol (Labs Concurrent Multiprotocol) that is not backward compatible with `Spinel` that is the standard for Open  thread. The good side is that Silab can running Zigbee and OTBR with one RCP firmware and have all apps on the host system. And also running EZSP on the SOC (as NCP) and OTBR on the host system with only RCP on the chip.
The last is described here https://github.com/home-assistant/addons-development/blob/master/silabs-concurrent-multiprotocol/DOCS.md.

The over all RCP is here https://github.com/home-assistant/addons/blob/master/silabs-multiprotocol/DOCS.md#architecture.

One more problem with Silabs implanting is that they is forcing encrypted communication between the SOC and the host system with the private keys in the firmware. For the moment we have disabling it in the firmware and the host application (i think we can doing one "open key for putting in our firmware and its more standard but is on later story).

You only need changing `CPC Security Configuration` to `Off` and configure the comport in the hardware (com0 shall being used) and if its possible in the hardware using  hardware flow control (software is not supported so use non as alternative).
Also its good using higher baud rate then 115200 if its possible then the 802.15.4 is max 265 plus over head in the protocol to the host system (not one must but recommended if the hardware is working OK with it).

Also is Silabs moving fast and making large changes every release of GSDK so need cooking new firmware for every new GSDK release.

I have my "Billy RCP" (IKEA EFR32MG1P) module running for many months with 3 OTBR and ZHA with Ziogbee network and its working better and better.
tuya	ZS3L Zigbee module (with EFR32MG21A020F768IM32-B) is also working great with RCP firmware.

I have not time testing NCP / RCP firmware and its npt possible on MG1 device then they is having to less flash for it and tuya ZS3L / HA Sky Connect is not so easy flashing as my "Billys" but is coming later.

Try doing one GSDK 4.2.0.0 RCP firmware and testing it with the HA addon its for the moment the best and easy way and is doing Zigbee / ZHA (Z2M shall working 2) but thread is only possible adding more border routers and not interface for adding other thread devices but is shall coming in HA soon if all is going well.

And great thanks for supporting our Zigbee / Thread community !!!!

xsp1989, 2022-12-20:
By the way, ZB-GW04 has updated the hardware version to v1.2, this version already supports CTS/RTS hardware flow control, and NCP and RCP firmware supporting hardware flow control will be released later

xsp1989, 2022-12-20:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RCPMultiPAN

I built a RCP firmware, so far I only tested that it can be added to "Silicon Labs Multiprotocol", I didn't test if I can add a device, and if it works fine. So it is not recommended that everyone upgrade for testing.

MattWestb, 2022-12-20:
Testing thread is best setting up one more HA with the add.on and joining the second one to the first that have forming its network. Then you shall getting 2 routers in the OTBR GUI topology.
Adding ZHA have being little tricky with timeouts and if having problem getting it working restart the add-on or HA and its normally working.
Important is then setting up ZHA is if changing chip its best forming on new network then the IEEE is changed and can making problems.

Now you only need getting Tasmota working with Matter :-)) (joke).

Great work done !!

PS: Its possible using network port for the RCP communication but is not so well tested and may not  working OK (only older version was fast tested but ESP chips dont support HA flow control).

skgsergio, 2022-12-20:
Thanks @xsp1989, I've just flashed my ZB-GW04 v1.1 based USB dongle with the RCP MultiPAN firmware (`rcp-uart_nsw_115200_v4.2.0.gbl`) and set it up in Home Assitant with the `Silicon Labs Multiprotocol` addon and even restored the backup of my previous network and looks like everything is working. Lets see if it is stable :)

Loic691, 2022-12-20:
Hi
What is difference between rcp and ncp firmware ?
Is it possible to test rcp multiprotocol on the sonoff ZB Dongle E ?
What is the protocol used after from Z2M ? EZSP ?

MattWestb, 2022-12-20:
RCP (Radio Co Processor) is described her: https://github.com/home-assistant/addons/blob/master/silabs-multiprotocol/DOCS.md#architecture.

Normally is the Zigbee stack running one the SOC (system on chip) and communicating with the host system over one communication port = NCP (Network Co Processor) and with the host system with one proprietary protocol like EZSP over one other com  port.

In RCP mode is only the radio living on the chip and one small program is tunneling the information to the host system over one com port.  On the host system is the Zigbee stack  (zigbeed) "living" and is doing all things and the radio command is sen and received to / from the radio chip over the comport and the host system is talking to the upper system with EZSP.

With RCP is possible using one radio chip for more protocols like Zigbee, Thread and Bluetooth then the host system is doing the heavy work. This firmware can being used for Zigbee and Open Thread Boarder Router on the host system.

Its also possible using Zigbee / NCP on the SOC and RCP for thread on the host system but its not wildly tested for the moment.

With RCP firmware and the Silicon Labs Multiprotocol addon you is getting one network connection for EZSP that is working with ZHA and shall working with Z2M 2 if all is OK plus one Open Thread Boarder Router in HA.

Sonoff ZB Dongle E shall working with  this firmware but is not recommended for normal user then its not widely tested.

And is its not working it can being problems flashing one normal firmware on the device. 

The protocol Silabs Zigbee is talking is EZSP and to Z2M its being tunneled over Ethernet as one com port.

Hedda, 2023-01-23:
FYI, Silicon Labs has now also released Gecko SDK 4.2.0 with Zigbee EmberZNet SDK 7.2.0.0

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.0

Any chance you could release some builds for zigpy and zigbee-herdsman developers to test?

It looks to use EZSP v9 (EmberZNet Serial Protocol version 9) same as Zigbee EmberZNet 7.1

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.0.0.pdf

New features are mostly related to Dynamic Multiprotocol and multi-PAN 802.15.4 in RCP mode:

Secure key storage support for MG2x parts that support Secure Vault-High

Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode

Dynamic Multiprotocol Zigbee-NCP + Bluetooth-NCP (experimental)

Manufacturing Library (MfgLib) support for Concurrent Multiprotocol RCP

Zigbee + OpenThread Concurrent Lis-tening on MG24 parts (experimental)

Zigbee on Host (ZigbeeD) support for 32 bit and 64 bit x86 architecture (experimental)

MGM240S SiP Module Support

They also added simultaneous Dual-Band and Dual-PHY for Zigbee Smart Energy on MG24/MG12:

MG24+Si4468 Dual-PHY Zigbee Smart Energy support
MG12 Dual-Band 2.4GHz + SubGHz Zigbee Smart Energy support
Deprecated in release 7.2.0.0:

The Secure EZSP feature will be removed in a future release.

spartandrew18, 2023-01-24:
BUMP. The multipan firmware bug has now been fixed. I was wondering is there anyway to add the zigbee configuration to zigbee2mqtt?

When I try to add the information below I get an error from the logs.

port: socket://core-silabs-multiprotocol:9999
adapter: ezsp

[17:13:38] INFO: Preparing to start...
[17:13:39] INFO: Socat not enabled
[17:13:41] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info 2023-01-24 17:13:48: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-24.17-13-48' filename: log.txt
Zigbee2MQTT:info 2023-01-24 17:13:48: Starting Zigbee2MQTT version 1.29.2-dev (commit #7639ce4)
Zigbee2MQTT:info 2023-01-24 17:13:48: Starting zigbee-herdsman (0.14.88)
Zigbee2MQTT:error 2023-01-24 17:13:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-24 17:13:48: Failed to start zigbee
Zigbee2MQTT:error 2023-01-24 17:13:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-24 17:13:48: Exiting...
Zigbee2MQTT:error 2023-01-24 17:13:48: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open socket://core-silabs-multiprotocol:9999'
at SerialPort. (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:91:28)
at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:76:22)
at /app/node_modules/@serialport/stream/dist/index.js:112:18

MattWestb, 2023-01-24:
Do you using:
```
serial:
    port: 'tcp://core-silabs-multiprotocol:9999'
```
as from the instruction https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_3-configure or the IP ?

The fixed is not the latest version its 7.1.4 but i think Silabs is releasing the more stable / bug fixed 7.2.1 very soon.

GSDK 4.2.0 = EZSP 7.2.0 and so on.

spartandrew18, 2023-01-24:
Using TCP instead of socket results in:

[18:14:52] INFO: Preparing to start...
[18:14:53] INFO: Socat not enabled
[18:14:56] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-01-24 18:15:01: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-24.18-15-01' filename: log.txt
Zigbee2MQTT:info  2023-01-24 18:15:01: Starting Zigbee2MQTT version 1.29.2-dev (commit #7639ce4)
Zigbee2MQTT:info  2023-01-24 18:15:01: Starting zigbee-herdsman (0.14.88)
Zigbee2MQTT:error 2023-01-24 18:15:02: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-24 18:15:02: Failed to start zigbee
Zigbee2MQTT:error 2023-01-24 18:15:02: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-24 18:15:02: Exiting...
Zigbee2MQTT:error 2023-01-24 18:15:02: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:147:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)

MattWestb, 2023-01-25:
Have you doing point [1](https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_1-install-ser2net) then you is getting:
```
[18:14:53] INFO: Socat not enabled
```` 
?

spartandrew18, 2023-01-25:
> Have you doing point [1](https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_1-install-ser2net) then you is getting:
> 
> ```
> 
> [18:14:53] INFO: Socat not enabled
> 
> ```` 
> 
> ?

I am unsure how to do point 1 on home assistant terminal since sudo would not work.

skgsergio, 2023-01-25:
The latest version of Home Assistant Silabs Multiprotocol Addon (0.12.0) switched from Gecko SDK 4.2.0 to 4.1.4, which cannot work with `rcp-uart_nsw_115200_v4.2.0.gbl` and fails to start.

I've started configuring and compiling the MultiPAN/Muliprotocol firmware myself for my ZB-GW04 v1.1 based USB dongle. I'm running the latest Home Assistant Silabs Multiprotocol Addon version (0.12.0) with the Gecko SDK 4.1.4 firmware I've uploaded here: https://github.com/skgsergio/silabs-multiprotocol-firmware-zbgw04-usb

Hedda, 2023-01-25:
> BUMP. The multipan firmware bug has now been fixed. I was wondering is there anyway to add the zigbee configuration to zigbee2mqtt?
> 
> When I try to add the information below I get an error from the logs.
> 
> port: socket://core-silabs-multiprotocol:9999 adapter: ezsp

 @spartandrew18  That is off-topic in this specific request issue as I here asked for "Zigbee in NCP mode and Open-Thread in RCP mode" and that means you can use Zigbee directly and do not need to use zigbeed.

The correct topic for "Multi-PAN RPC for Zigbee in RCP mode and Open-Thread in RCP mode" is this other thread -> https://github.com/xsp1989/zigbeeFirmware/issues/21

For Zigbee2MQTT specifically however, it is better if you ask in their project -> https://github.com/Koenkk/zigbee2mqtt/issues/14497

MattWestb, 2023-01-25:
Hola @skgsergio i was thinking doing the same but i dont have the device so i cant testing if its working or not.
Dont forgetting putting in CTUNE value = 128 (i have not looking where doing it in SS5)
 or you is getting problems with miss tuned radio and end devices cant connecting OK.

MattWestb, 2023-01-25:
@skgsergio In SS5 with GSDK 4.1.4 in your project and software components > `Services` > `Device Init: HFXO`. Configure and setting `CTUNE` to what you need.

skgsergio, 2023-01-25:
@MattWestb thanks, will change it and upload new versions.

MattWestb, 2023-01-25:
Dont forgetting testing if sleeping end devices can staying connected in the log run.
I using IKEA MG1P modules and 4.2.0 is working OK if not having end device connected to the RCP (i have blocking it having direct children) and was working OK with earlier RCP 2 but was having more problems that is fixed now.

GREAT WORK DONE !!

skgsergio, 2023-01-25:
I've reuploaded both v4.2.0 and v4.1.4 MultiPAN RCP firmware builds with CTUNE=128. I've tested both on my device.

Anyways I had it running with the default CTUNE (140) with my v4.1.4 build all night and both my ZigBee motion sensors (Philips Hue SML001) were working this morning. But checking Home Assistants graph they seem to be connected to other devices (a bulb and a plug).

MattWestb, 2023-01-25:
Great !!
The HUE motion sensor is is one semi sleeper and shall not having great problems (more then its having = its jumping to bad routers in my production system until its finding the worse parent and then staying on that). IKEA controllers and HUE remotes was daring batteries before Sonoff was fixing the c-tune and after that is better but its bugs in the device firmware that is ding that.
Also paring can being tricky if the radio is miss tuned but it shall no being one problem now.

And the c-tune is only helping for devices that have direct link with the RCP so your was having one router between = working for 100% if its one good router.

fhloston, 2023-03-21:
Can the multiprotocol firmware be produced for the SM-011 module in the ZB-GW03?

How is the signature done and enforced?

I am trying to find out, if the firmware can also be built here https://github.com/ksjh/silabs-firmware-builder/issues/8

Hedda, 2023-03-21:
> Can the multiprotocol firmware be produced for the SM-011 module in the ZB-GW03?

Please start a separate thread for that since your question is about a signed firmware image for a locked/closed product.

@xsp1989 have provided signed firmware images for ZB-GW03 in the past but I think that your question about getting a firmware image signed is different from the original request is this issue.

fhloston, 2023-03-21:
> Please start a separate thread for that since your question is about a signed firmware image for a locked/closed product.

Ok, can do that. For now I was trying to understand requirements and limitations of the SM-011 in the ZB-GW03 specifically.

xsp1989, 2023-03-22:
The signature of ZB-GW03 cannot be turned off unless a bootloader without signature is refreshed through jlink. Currently, the signature key is a company secret and will not be provided to the public. If necessary, I will provide a signed firmware next week (if possible correctly signed and run).

closed 2024-01-04 by xsp1989