xsp1989/zigbeeFirmware #33: [REQUEST] Silabs Zigbee EmberZNet SDK 7.2 NCP experimental builds for testing?

Hedda, 2023-01-23:
FYI, Silicon Labs has now also released Gecko SDK 4.2.0 with Zigbee EmberZNet SDK 7.2.0.0

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.0

Any chance you would consider releasing some experimental builds for [zigpy](https://github.com/zigpy/zigpy/discussions) and [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/issues/319) developers to test?

It looks to use EZSP v9 (EmberZNet Serial Protocol version 9) same as Zigbee EmberZNet 7.1

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.0.0.pdf

Feature news are mostly related to Dynamic Multiprotocol and multi-PAN 802.15.4 in RCP mode:

- Secure key storage support for MG2x parts that support Secure Vault-High

- Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode
- Dynamic Multiprotocol Zigbee-NCP + Bluetooth-NCP (experimental)
- Manufacturing Library (MfgLib) support for Concurrent Multiprotocol RCP
- Zigbee + OpenThread Concurrent Lis-tening on MG24 parts (experimental)
- Zigbee on Host (ZigbeeD) support for 32 bit and 64 bit x86 architecture (experimental)
- MGM240S SiP Module Support

They also added simultaneous Dual-Band and Dual-PHY for Zigbee Smart Energy on MG24/MG12:

- MG24+Si4468 Dual-PHY Zigbee Smart Energy support
- MG12 Dual-Band 2.4GHz + SubGHz Zigbee Smart Energy support

Deprecated in release 7.2.0.0:

- The Secure EZSP feature will be removed in a future release.

PS: Silicon Labs also released Gecko SDK (GSDK) Version 4.1.4.0 as a bug-fix release that contains Zigbee EmberZNet SDK 7.1.4.0:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.4

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.4.0.pdf

xsp1989, 2023-01-27:
The new experimental version may take a while to be released, and you can also build it yourself. Currently, the latest version of GSDK can be installed and built directly without SN, or even logging in to the account of Simplicity Studio.

ashimokawa, 2023-01-28:
I tried to build an ncp application firmware, but I did not get how to get how to get the NCP UART SW example without SimplicityStudio. I downloaded and installed a working gcc/gecko_sdk/slc_cli combination, but, well, no examples that seems to fit.

"Silicon Labs recommends that you use either the NCP SPI, NCP UART (HW), or NCP UART SW example applications as a starting point for building a customized NCP application."

The gecko sdk only has ./protocol/zigbee/app/ncp/sample-app/mp-ncp-uart-hw nothing with -sw

Is the firmware in this repo built from the vanilla sample app? If so which one? Is it accessible without SimplicityStudio?

MattWestb, 2023-01-28:
Install Simplicity Studio 5 and use the example for for NCP UART for your chip / module and you can configuring all in the GUI an then building the firmware.

ashimokawa, 2023-01-28:
@MattWestb 

Thank you for your reply.

I installed SimplicityStudio 5 and I see the same examples as before, just in a GUI, I do not understand why Silicon Labs documents always refer to a "NCP UART SW" when there is only a HW example. I will try to figure it out but I have a hunch I am on the wrong path here.

I tried with the 4.1.4 and 4.2.0 sdk releases.

MattWestb, 2023-01-28:
NCP have SW implanted but OT RCP is only having HW or non = bad.
Easiest is pin tool and config pins for one comport and then clicking on the software component and setting the com parameters.

ashimokawa, 2023-01-28:
@MattWestb 

Thanks, got it. And built. I did not flash but at least now I understood that the newer SDKs just do not have the -sw example and I have to click my way through to switch it sw flow control.

xsp1989, 2023-01-29:
[image]
@ashimokawa In the flow control setting, select CTS/RTS, and then select the corresponding GPIO in UART.

ashimokawa, 2023-01-29:
@xsp1989 

Thank you, this confirms what I already did.

For the other values (CTUNE, source routing tables), I edited header files, because I could not find them in the GUI.

MattWestb, 2023-01-29:
I SS5 components:
C-Tune:
Services
Device Initialization
Device Init: HFXO
CTUNE

Source routing:
Zigbee
Stack
Source route
Zigbee Source Route Library configuration

xsp1989, 2023-01-30:
https://drive.google.com/drive/folders/14f8lY3HvRxh_B4Q4ZmVvLs5At1Qc1BZp
This is the schematic diagram of ZB-GW04

MattWestb, 2023-01-30:
Thanks one more time !!

hitokiri8x, 2023-02-01:
I only tested the firmware for 1 day on ZBDongle-E 3.0 I can say that on my adapter is better than stock one, at least my aqara devices were still online and the application survived a reboot. ( on stock firmware they disconnect after some hours ).
Sadly as I've already initiated the return procedure I'm done testing and returning the device, people with same device keep looking in this firmware, is promising

Hedda, 2023-04-17:
FYI, the latest bug-fix version from Silicon Labs is now Zigbee EmberZNet SDK 7.2.2.0 from Gecko SDK (GSDK) Version 4.2.2.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.2.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.2