xsp1989/zigbeeFirmware #34: Hardware settings of ZB-GW04

MattWestb, 2023-01-25:
Sorry for pinging you @xsp1989 then you is not working for Sonoff longer.
I have on technical question. You was saying that the ZB-GW04 is having PCB lines for hardware flow control but i was not asking for the pin / pads used in both ends.
We have one user that is trying (successfully) cooking RCP firmware for that hardware and its better if we can getting the HW flow control working for getting stable communication with the RCP.

If you have time and knowledge your replay is more then welcome and if not its very OK.

All the best for you !!

Mattias W

xsp1989, 2023-01-27:
ZB-GW04 is designed and produced by easyiot, CTS/RTS is only supported by the latest hardware version, and the hardware version is v1.3.

MattWestb, 2023-01-27:
Is you knowing and can sharing witch pins ( pads is using for CTS / RTS ?

Thanks in advance.

xsp1989, 2023-01-27:
[image]

PD03-RTS
PD04-CTS

MattWestb, 2023-01-27:
Great thanks also for all support you have done for the community !!

Mattias W

closed 2023-01-27 by MattWestb