xsp1989/zigbeeFirmware #39: EmberZNet v7.2.2.0 firmware

mamrai1, 2023-03-22:
Can you please add EmberZNet v7.2.2.0 firmware for testing?

Hedda, 2023-04-17:
This is really a duplicate of https://github.com/xsp1989/zigbeeFirmware/issues/33 which is meant to be for all Zigbee EmberZNet 7.2.x.x as those are bug-fix releases of 7.2 releases.

xsp1989, 2023-04-17:
I built a 7.2.2.0 software flow control firmware, but I didn't see the software flow control message, restarted the NCP, and it directly output "1A C1 02 02 9B 7B 7E", the previous firmware would output "11 1A C1 02 02 9B 7B 7E ", do you understand this question?

frfstu, 2023-04-18:
I'm update dongle ZB-GW04 v1.2 up to firmware version https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.2.2.0_115200.gbl using Elelabs_EzspFwUtility:

2023/04/18 15:47:07 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2023/04/18 15:47:07 Elelabs_EzspFwUtility:   Firmware: 6.10.3-41
2023/04/18 15:47:07 Elelabs_EzspFwUtility:   EZSP v8
2023/04/18 15:47:07 Elelabs_EzspFwUtility:   Launch in bootloader mode
2023/04/18 15:47:16 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2023/04/18 15:47:16 Elelabs_EzspFwUtility:   Gecko Bootloader v1.9.2
2023/04/18 15:47:17 Elelabs_EzspFwUtility:   Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
...
2023/04/18 15:47:50 Elelabs_EzspFwUtility:   Firmware upload complete
2023/04/18 15:47:50 Elelabs_EzspFwUtility:   Rebooting NCP...
2023/04/18 15:47:58 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2023/04/18 15:47:58 Elelabs_EzspFwUtility:   Firmware: 7.2.2-190
2023/04/18 15:47:58 Elelabs_EzspFwUtility:   EZSP v11

frfstu, 2023-04-18:
2023-04-18 18:40:45.452 [WARN ] [mmon.WrappedScheduledExecutorService] - Scheduled runnable ended with an exception: 

java.lang.NullPointerException: null

	at com.zsmartsystems.zigbee.dongle.ember.EmberNcp.getConfiguration(EmberNcp.java:457) ~[?:?]

	at com.zsmartsystems.zigbee.dongle.ember.internal.EmberStackConfiguration.getConfiguration(EmberStackConfiguration.java:72) ~[?:?]

	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(ZigBeeDongleEzsp.java:439) ~[?:?]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:418) ~[?:?]

	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:431) ~[?:?]

	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$2(ZigBeeCoordinatorHandler.java:557) ~[?:?]

	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]

	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]

	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]

	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]

	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]

	at java.lang.Thread.run(Thread.java:829) [?:?]

frfstu, 2023-04-18:
openHABian Configuration Tool — [main]{2023-04-13T22:49:27+02:00}(986ec5e)                                       │ 
                              │ openHAB 3.4.3 - Release Build

mamrai1, 2023-04-18:
I flashed sonoff zbdongle plus with 7.2.2.0 and it works with z2m... 
[image]

frfstu, 2023-04-18:
2023-04-18 18:41:45.696 [WARN ] [nding.zigbee.serial.ZigBeeSerialPort] - Processing DATA_AVAILABLE event: received IOException in serial port event

java.io.IOException: Expected to be able to read 10 bytes, but saw error after 0

	at org.openhab.binding.zigbee.serial.ZigBeeSerialPort.serialEvent(ZigBeeSerialPort.java:343) [bundleFile:?]

	at org.openhab.core.io.transport.serial.rxtx.RxTxSerialPort$1.serialEvent(RxTxSerialPort.java:82) [bundleFile:?]

	at gnu.io.RXTXPort.sendEvent(RXTXPort.java:834) ~[bundleFile:5.2.1.OH1]

	at gnu.io.RXTXPort.eventLoop(Native Method) [bundleFile:5.2.1.OH1]

	at gnu.io.RXTXPort$MonitorThread.run(RXTXPort.java:108) [bundleFile:5.2.1.OH1]

frfstu, 2023-04-18:
a similar situation is described in the logs here https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjByLWT67P-AhW6BhAIHeg5AikQFnoECA0QAQ&url=https%3A%2F%2Fcommunity-openhab-org.s3.dualstack.eu-central-1.amazonaws.com%2Foriginal%2F3X%2Fd%2Fb%2Fdbdd17c6364ea7095f9e85ded2009489fdb0d815.log&usg=AOvVaw1pXYNgoAPfNP49vE0NQtUh

frfstu, 2023-04-18:
##   Release = Raspbian GNU/Linux 11 (bullseye)
##    Kernel = Linux 6.1.21-v7+
##  Platform = Raspberry Pi 3 Model B Plus Rev 1.3
##    Uptime = 0 day(s). 0:1:42
## CPU Usage = 91.86% avg over 4 cpu(s) (4 core(s) x 1 socket(s))
##  CPU Load = 1m: 2.40, 5m: 0.88, 15m: 0.32
##    Memory = Free: 0.33GB (35%), Used: 0.61GB (65%), Total: 0.94GB
##      Swap = Free: 1.80GB (100%), Used: 0.00GB (0%), Total: 1.80GB
##      Root = Free: 95.14GB (86%), Used: 14.24GB (14%), Total: 114.06GB
##   Updates = 0 apt updates available.
##  Sessions = 1 session(s)
## Processes = 130 running processes of 32768 maximum processes
###############################################################################

                          _   _     _     ____   _               
  ___   ___   ___   ___  | | | |   / \   | __ ) (_)  ____   ___  
 / _ \ / _ \ / _ \ / _ \ | |_| |  / _ \  |  _ \ | | / _  \ / _ \ 
| (_) | (_) |  __/| | | ||  _  | / ___ \ | |_) )| || (_) || | | |
 \___/|  __/ \___/|_| |_||_| |_|/_/   \_\|____/ |_| \__|_||_| | |
      |_|                  openHAB 3.4.3 - Release Build

Hedda, 2023-04-18:
> openHAB not detect firmware version

@frfstu openHAB ZigBee Binding still only support EZSP v8 (EmberZNet Serial Protocol version 8) so I believe that latest firmware that you can use is EmberZNet 7.0 (7.0.2).

FYI, the Zigbee framework library that the openHAB ZigBee Binding depends on did however get updated to support EZSP v10 support about a month ago, so it will probably not be long before the openHAB ZigBee Binding update its dependencies with that newer library, but even then I believe it will might not be able use EmberZNet 7.2.2.2 firmware as that was updated to EZSP v11 which that library that openHAB ZigBee Binding depends on do not yet support.

MattWestb, 2023-04-18:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1378

Hedda, 2023-04-18:
> zsmartsystems/com.zsmartsystems.zigbee#1378

Yes EZSP v10 support is already upstream library that depends on but it does not look like the OpenHAB ZigBee Binding has bumped its dependencies to use that yet? -> https://github.com/openhab/org.openhab.binding.zigbee/

Hedda, 2023-05-05:
> Can you please add EmberZNet v7.2.2.0 firmware for testing?

FYI, Silabs Zigbee EmberZNet SDK	7.2.3.0 has now been released as part of Silicon Labs Gecko SDK (GSDK) Version 4.2.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.3.0.pdf

Silicon Labs Zigbee EmberZNet 7.2.3.0 release states that it is specifically targeting quality improvements and bug fixes.

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.3

xsp1989, 2023-05-05:
> FYI, Silabs Zigbee EmberZNet SDK 7.2.3.0 has now been released as part of Silicon Labs Gecko SDK (GSDK) Version 4.2.3.0 [...]

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

The new firmware has been released, but I need to declare that this firmware has not been strictly tested, it is only rebuilt after upgrading to the latest SDK, and no parameters have been changed compared to the previous version

mamrai1, 2023-05-07:
7.2.3.0 firmware doesn't work with z2m...

2023-05-07T08:00:00.631695467Z 2023-05-07T08:00:00.631Z zigbee-herdsman:adapter:ezsp:erro zdoRequest error
2023-05-07T08:00:00.631950411Z 2023-05-07T08:00:00.631Z zigbee-herdsman:adapter:ezsp:debg Node descriptor request for '0' failed (Error: ZdoRequest error), retry
2023-05-07T08:00:00.720815334Z Error: ZdoRequest error
2023-05-07T08:00:00.720867427Z     at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:573:19)
2023-05-07T08:00:00.720883812Z     at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:373:28)
2023-05-07T08:00:00.720893574Z     at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:363:32)
2023-05-07T08:00:00.720902181Z     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Hedda, 2023-05-07:
@mamrai1 as explained above to @frfstu please understand and remember that you need to report any issues seen in Zigbee gateway application(s) to the projects that is developing that said Zigbee gateway application(s), not here, regardless of which Zigbee hardware adapter and firmware version that you have flashed. It is not the same developers that are releasing firmware images that are working on the many different Zigbee gateway applications that can use it. The firmware images for Zigbee Coordinator adapters that you can find here can be used by any Zigbee gateway applications, and there Zigbee2MQTT, IoBroker, ZHA integration, for Home Assistant, Zigbee plugin for Jeedom, Zigbee plugin for Domoticz, are just a few of those. Any issues with different firmware versions will need to be fixed by the developera of those applications.

mamrai1, 2023-05-07:
> @mamrai1 as explained above to @frfstu and clearified further in #44
> 
> Please understand and remember that you need to report any issues seen in Zigbee gateway application(s) to the projects that is developing that said Zigbee gateway application(s), not here, regardless of which Zigbee hardware adapter and firmware version that you have flashed.
> 
> It is not the same developers that are releasing firmware images that are working on the many different Zigbee gateway applications that can use it.
> 
> The firmware images for Zigbee Coordinator adapters that you can find here can be used by any Zigbee gateway applications, and there Zigbee2MQTT, IoBroker, ZHA integration, for Home Assistant, Zigbee plugin for Jeedom, Zigbee plugin for Domoticz, are just a few of those.
> 
> Any issues with different firmware versions will need to be fixed by the developers of those applications, not by the people buildibg and releasing firmware images.

I reported the problem here because xsp1989 has not tested the firmware....

mamrai1, 2023-05-07:
@xsp1989 Maybe some of the changed parameters you did last week on latest 7.1.4.0 and 7.2.2.3 are causing the issue with z2m?

Hedda, 2023-05-07:
> I reported the problem here because xsp1989 has not tested the firmware....

That is exactly the point. The person who builds and releases Zigbee firmware images is not expected to test the firmware in all Zigbee gateway applications that it might be compatible with. It is instead up to the community of each separate Zigbee gateway application to test and report issues to their developers, (and again not reporting Zigbee gateway application to the person who is just building and releasing. Zigbee firmware images).

I now read that you have reported your issues with Zigbee2MQTT to the developers of zigbee-herdsman, and that is great!

I mean no offence, I just want to again want to clearly that you should really not report issues or problems with any Zigbee gateway applications here at all, instead only report issues and problems to just the developers of the Zigbee gateway application that you use. It is not possible to deal with any Zigbee gateway applications here and it just spamming if everyone keeps reporting all issues with new firmware seen in all Zigbee gateway applications that it could be compatible with.

> Maybe some of the changed parameters you did last week on latest 7.1.4.0 and 7.2.2.3 are causing the issue with z2m?

[xsp1989](https://github.com/xsp1989) wrote above; _"I need to declare that this firmware has not been strictly tested, it is only rebuilt after upgrading to the latest SDK, and **no parameters have been changed compared to the previous version**"_

xsp1989, 2023-05-08:
What I said has not been strictly tested means that it has not passed long-term stability tests, access to various platforms, and access to a large number of different devices. Before the firmware is released, we will access ZHA or Z2M to test the access of some devices and control them. Normally, siliconLabs will strictly test the SDK. We just use the official demo to modify the parameters and rebuild it. This kind of serious bug will not appear. As @Hedda said, for each different platform, our firmware cannot be universally adapted.

By the way, thanks to @Hedda for the support.

xsp1989, 2023-05-08:
@Hedda For EFR32MG21A020, this chip supports a maximum transmit power of 20dbm, but the firmware defaults to a maximum transmit power of 10dbm. On the host side, you may use the [Name: setRadioPower ID: 0x0099] command to modify the maximum transmit power. Of course, our firmware has already used a maximum transmit power of 20dbm

[image]

mamrai1, 2023-05-08:
@xsp1989 can you please confirm speed in ncp-uart-sw_7.2.3.0_115200.gbl and latest sw_7.1.4.0 is set to 115200?

xsp1989, 2023-05-08:
If you suspect that there is a problem with the baud rate, you can do the following, use a serial port tool to open the USB serial port, set the baud rate to 115200, set it to HEX receiving mode, and then press RST on the dongle, you will see One output "1A C1 02 02 9B 7B 7E"

juanchula, 2023-06-04:
@xsp1989 Firmware 7.2.3-215 return 02 02 9b 7b 7e 1a c1

closed 2024-02-28 by xsp1989