xsp1989/zigbeeFirmware #41: [SUGGESTION] SL Web Tools implementation as Web Flasher for flashing Silicon Labs EFR32 firmware via a web browser

Hedda, 2023-04-13:
Home Assistant founders (and Nabu Casa) have recently among other things announced: "[SL Web Tools](https://github.com/NabuCasa/sl-web-tools)" project (short for "Silicon Labs Web Tools") is available as open source, so wondering if you would consider adding that to your website as a tool for updating/flashing easyiot ZB-GW04 adapters/dongles based on Silicon Labs EFR32 SoC/MCU family. See:

https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/#silabs-multi-flasher--sl-web-tools

(Note that this is only just one section from https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/ )

SL Web Tools allows users to manage the firmware on Zigbee/Thread sticks directly from a web browser, with no terminals or compilation necessary. This allows product creators like yourself to offer easy firmware updates to their users and make experimental firmware updates available to test bug fixes quickly.

SL Web Tools runs in the browser using [Pyodide](https://pyodide.org/en/stable/) and is currently powered by another new open-source Python project from them, “[silabs-universal-flasher](https://github.com/NabuCasa/universal-silabs-flasher)” for Silicon Labs EFR32 based SoCs like the EFR32MG21 SoC.

https://github.com/NabuCasa/sl-web-tools

(and currently powered by https://github.com/NabuCasa/universal-silabs-flasher for Silicon Labs based chips)

https://www.youtube.com/watch?v=-88K23e8XYw&t=2s&ab_channel=HomeAssistant

To see it in action, a demo of SL Web Tools for Silicon Labs chips has been added to the Home Assistant SkyConnect documentation.

https://skyconnect.home-assistant.io/firmware-update/ redirected from https://skyconnect.home-assistant.io/ 

[image]

Nabu Casa employee puddy very recently also added support for flashing Thread/OpenThread firmware with sl-web-tools:

https://github.com/NabuCasa/sl-web-tools/pull/7

Also, if wanted to take it to the next level then maybe forking that for easyiot ZB-GW04 Silabs based adapters/dongles by starting a new project called "TI Web Tools" (short for "Texas Instruments Web Tools") or similar that is based on that in combination with [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl) code (and/or [ZigStarGW-MT](https://github.com/xyzroe/ZigStarGW-MT)) for the TI flashing part(?). That is, an idea to extend this would be to replace the [silabs-universal-flasher](https://github.com/NabuCasa/universal-silabs-flasher) code part with [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl) and/or [ZigStarGW-MT](https://github.com/xyzroe/ZigStarGW-MT) code in order to allow flashing of Texas Instruments based adapters instead.

Note! [SL Web Tools](https://github.com/NabuCasa/sl-web-tools) is in turn a fork of their previous "[ESP Web Tools](https://esphome.github.io/esp-web-tools/)" project which allow DIY firmware flashing of ESP32/ESP8266 SoC projecst to offer an easy web installer on their homepage, allowing users to get started directly from their browser. It has been adopted by all the great projects, including [Tasmota](https://tasmota.github.io/install/), [WLED](https://install.wled.me/), and [ESPresense](https://espresense.com/firmware). Ever since the introduction of ESP Web Tools, they expanded this to more devices, and now they also introduced [SL Web Tools](https://github.com/NabuCasa/sl-web-tools) for SiLabs based chips.

PS: For reference; the firmware build system for Silicon Labs was made by Home Assistant developer/engineer and Nabu Casa employee [Stefan Agner (a.k.a. agners](https://github.com/agners), who also developed the EFR32MG21-based [Home Assistant SkyConnect USB stick/dongle](https://www.home-assistant.io/skyconnect) as well as the EFR32MG21-based radio module integrated into the Home Assistant Yellow appliance for Rasberry Pi 4 Compute Module). And the "universal-silabs-flasher" firmware tool was made by [puddly](https://github.com/puddly) who is also one of the lead developers of [zigpy ](https://github.com/zigpy/zigpy)and the [radio libraries for zigpy](https://github.com/zigpy) that is used in Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) (among other projects).

Hedda, 2023-04-13:
FYI, @darkxst has published a barebone implementation hosted on Github pages for easyiot ZB-GW04 as proof-of-concept:

https://github.com/darkxst/sl-test/tree/gh-pages

https://darkxst.github.io/sl-test/

He does however write that it would, however, require some modifications to the build system:

https://github.com/ksjh/silabs-firmware-builder/issues/12

That in turn depends on silabs-firmware-builder fork from ksjh here:

https://github.com/ksjh/silabs-firmware-builder

xsp1989, 2023-04-27:
The 7.1.4.0 firmware has been re-released now, based on GSDK4.1.4. If you definitely need the 7.1.1.0 firmware, please go to the git commit history to find it.