xsp1989/zigbeeFirmware #42: where is firmware 7.1.1?

mamrai1, 2023-04-20:
i think firmware 7.1.1 is one of the most stable firmwares used with z2m. Why is not listed anymore?

frfstu, 2023-04-21:
https://github.com/xsp1989/zigbeeFirmware/blob/edf2a0d3a717373764ec0673d4fecb0b882a636a/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.1.1.0_115200.gbl

I absolutely agree with you, today I installed this version on a recently purchased usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202110211-if00 and I confirm the stability of its work with z2m, unlike version 7.2.2 in which my devices turned out to be incompatible (previously working properly for more than three years) https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1 and https://www.zigbee2mqtt.io/devices/ZSS-ZK-THL.html#moes-zss-zk-thl. Separately, I want to note that in my openHAB 3.4.3 system on Raspberry Pi 3B + I was not able to achieve full-fledged work with openHAB ZigBee Binding, which still refuses to determine the software version and mac address even Dongle from SONOFF in firmware versions later than 6.10. which this Dongle is supplied with and with which ZigBee Binding works, but zigbee2MQTT does not work) which makes it impossible to use it with ZigBee Binding in versions 7.0.2, 7.1.1, 7.2.2 and apparently this state of affairs does not bother anyone , neither the developers of openHAB ZigBee Binding, nor SONOFF, because both of them declare full support and compatibility, indicating this in their descriptions, and even a "button" to add Ember Coordinator
SiLabs Ember based NCP is, that's just the result of this addition Status: OFFLINE COMMUNICATION_ERROR! But, although this is terribly inconvenient, inferior (with the complete absence of the possibility of creating automation), you will still be able to turn on / off devices and monitor the indicators displayed by them on the Zigbee2MQTT page in your browser, which in itself (even in such a " shell-shocked smart home) is far better than your waste-consuming collection of zigbee devices. I did not want to offend anyone, I am the happy owner of three Dongle's from different manufacturers, none of which work fully functional (at the same time, two manufacturers assure full compatibility in their descriptions)! And where is the zigbee community looking at all?! In general, something like this.

digiblur, 2023-04-21:
The CC2652 chipset is solid with Zigbee2mqtt

frfstu, 2023-04-21:
> The CC2652 chipset is solid with Zigbee2mqtt

https://docs.jethome.ru/ru/zigbee/sticks/zb6ux.html#zigbee-zb6ux-cc2652p1 I have been working wonderfully and fully in Home Assistant for almost four years!

xsp1989, 2023-04-22:
We have rearranged and re-released all EZSP firmware, optimized the default transmit power of the RF, but have not rebuilt 7.1.1, if you need, you can clone the entire project to the local, check out the old version with git, we The firmware version 7.1.1 will be re-released later, please understand the inconvenience caused to you.

xsp1989, 2023-04-27:
The 7.1.4.0 firmware has been re-released now, based on GSDK4.1.4. If you definitely need the 7.1.1.0 firmware, please go to the git commit history to find it.

closed 2023-07-04 by xsp1989