xsp1989/zigbeeFirmware #43: Request router FW for Sonoff Zigbee 3.0 USB Dongle Plus (Model "ZBDongle-E" with EFR32MG21 chip)

tomtzeng, 2023-04-21:
current router FW efr32mg21_zigbee_router_6.7.10.gbl can't work on Sonoff Zigbee 3.0 USB Dongle Plus (Model "ZBDongle-E" with EFR32MG21 chip)
have any chance to support it? event use new SDK version for router firmware?

xsp1989, 2023-04-22:
Although ZB-GW04 has similar hardware to Sonoff Zigbee 3.0 USB Dongle Plus, we cannot guarantee that our firmware can run normally on Sonoff’s dongle, so you can use Sonoff’s open firmware, we will not provide dedicated firmware for Sonoff support

tomtzeng, 2023-04-22:
> Although ZB-GW04 has similar hardware to Sonoff Zigbee 3.0 USB Dongle Plus, we cannot guarantee that our firmware can run normally on Sonoff’s dongle, so you can use Sonoff’s open firmware, we will not provide dedicated firmware for Sonoff support

Got it. Thanks.

closed 2023-04-22 by tomtzeng