xsp1989/zigbeeFirmware #46: [REQUEST] Silabs Zigbee EmberZNet SDK 7.3 NCP experimental builds for testing?

Hedda, 2023-06-19:
Silicon Labs have on the 7th of June 2023 released Gecko SDK (GSDK) v4.3 / 4.3.0 with Zigbee EmberZNet SDK v7.3 / 7.3.0.0 GA:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.0.0.pdf

https://www.silabs.com/wireless/zigbee?tab=software#software

Any chance would consider releasing some experimental builds if zigpy and zigbee-herdsman developers want to test EmberZNet 7.3?

Reference -> https://github.com/zigpy/bellows/issues/563 and https://github.com/Koenkk/zigbee-herdsman/issues/319

While brand new and not yet supported in any open-source Zigbee gateway solution, early cutting-edge can help pre-release testing.

Normal end-users should be warned against flashing this firmware was is the EZSP protocol version been incremented to EZSP v12 and Zigbee gateway solutions such as Home Asssitant's ZHA integration and Zigbee2MQTT are still only compatible with EZSP v11 and earlier EZSP versions at this time.

EZSP protocol version been incremented to EZSP v12 in Silicon Labs Gecko SDK v4.3 (4.3.0.0) / Zigbee EmberZNet v7.3 (7.3.0.0):

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.3/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

0x0C hexadecimal = 12 in decimal.

### Sidenote - EmberZNet 7.3 is the first Silabs firmware release to be "Zigbee PRO 2023" compatible

Silicon Labs EmberZNet 7.3.0.0 is the first Silabs SDK that offers Zigbee R23 compliance, with it making their Silabs Zigbee stack compatible with Zigbee Protocol Specification Revision 23 / R23 (also known as "Zigbee PRO 2023" and "Zigbee 2023" ).

**Zigbee**
- Zigbee R23 compliance, with these Security enhancements among others:
  - Dynamic link key negotiation
  - Device interview to query devices before they are allowed to join
  - Trust Center Swap Out to replace an existing Trust Center with a new one
  - Frame Counter Synchronization
- Zigbee Direct Device (ZDD) support for:
  - Onboarding/commissioning
  - Communication to all Zigbee devices without a hub (Alpha), using Bluetooth LE
- Zigbee Smart Energy 1.4a compliance (Alpha)
- Enhancements to Zigbee Green Power (ZGP) APIs
- New Zigbee Security upgrade component for moving encryption keys from cleartext NVM3 tokens into secure storage

https://www.silabs.com/blog/zigbee-pro-2023-spec-released-increases-security

https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

Reference:

https://github.com/zigpy/zigpy/issues/1211

Hedda, 2023-06-20:
Heads-up! Default flow control for Openthread changed in v4.3.0 from Hw to HwUart -> https://github.com/NabuCasa/silabs-firmware-builder/pull/18

Hedda, 2023-08-18:
FYI, Zigbee EmberZNet 7.3.1.0 SDK has been released as part of Gecko SDK (GSDK) Version 4.3.1.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.1.0.pdf

xsp1989, 2023-08-19:
Sorry, due to some special reasons, there is no time to update the firmware. Now the latest firmware 7.3.1.0 has been released, and the old SDK6.10.7.0 has also been updated

closed 2023-09-19 by xsp1989