xsp1989/zigbeeFirmware #48: 7.x firmware for GW03 hub / ZigbeeBridge_SM-011-signed ?

gdanov, 2024-01-06:
Hi,
 How can I convert the `.gbl` firmwares to `.ota` so that I can experiment with the GW03 hub? Given the chip inside is the same, is there objective reason we lack `.ota` versions for it?

MattWestb, 2024-01-06:
One NCP (coordinator) can only being updated with EBL/GBE files then its have one serial bootloader and devices can normally only being updated with OTA files with app bootloader (its depends how the firmware is being made) so its no user making OTA files for your device and if its signed its not possible doing it without the privet key we dont  have.

xsp1989, 2024-01-06:
ZB-GW03 uses an encrypted and signed bootloader. The ota file is just a rename. If necessary, I may release a 7.x version of the signed firmware, but this firmware may not be used by tasmota and can only be used Access Z2M or ZHA through TCP Bridge.

MattWestb, 2024-01-06:
The 7.3.X is working OK but still having problems so wait ti it more stable (we have starting running 7.4.0.0 ad RCP and its the same in the ground as the NCP version only different compiler and bugs).

gdanov, 2024-01-06:
> ZB-GW03 uses an encrypted and signed bootloader. The ota file is just a rename. If necessary, I may release a 7.x version of the signed firmware, but this firmware may not be used by tasmota and can only be used Access Z2M or ZHA through TCP Bridge.

My device is on esphome right now in TCP mode. So, for me new release would be possible to install & use. If you say there are known bugs and instabilities, then probably I'd rather pass because I don't want to re-pair my devices after the upgrade and subsequent rollback. I read backup & restore is possible but haven't tested yet.

In first place curious as I see such firmware is available for the USB stick with the same chip (and again used by z2m & zha), but not for the hub that uses same zb chip.

xsp1989, 2024-01-07:
ZB-GW03 and ZB-GW04 use different bootloaders. ZB-GW04 does not require a signature and can use any firmware without being affected.

brandonb1987, 2024-07-05:
Which is the latest working EZSP firmware for Tasmota then? Currently using your 14.0.0 build and 6.7.10

xsp1989, 2024-07-05:
> Which is the latest working EZSP firmware for Tasmota then? Currently using your 14.0.0 build and 6.7.10

Tasmota only supports EZSP V8. The latest version available is 6.10.9. You can also use 6.7.10. There will be no difference.

closed 2024-07-26 by xsp1989