xsp1989/zigbeeFirmware #49: EmberZNet 7.3.2.0/7.4.0.0 firmware

estimadarocha, 2024-02-08:
@xsp1989 do you expect to compile new version with 7.3.2 or 7.4.0?

Just a small apart dunno if this is the correct place to post it... this new versions working together with last z2m should support completely zgp devices?

Thanks

Hedda, 2024-03-19:
@xsp1989 +1 but for newer Zigbee EmberZNet 7.3.3.0 (GSDK 4.3.3.0) and EmberZNet 7.4.1.0 (GSDK 4.4.1.0) respectivly:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.3

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.3.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.1

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.4.1.0.pdf

MattWestb has summarized the included fixes for Zigbee EmberZNet NCP and RCP here:

https://github.com/zigpy/zigpy/discussions/894#discussioncomment-8782766

https://github.com/zigpy/zigpy/discussions/894#discussioncomment-8686629

He mentions there that his guess is EmberZNet 7.3.3.0 might be the last bug fix for and then it will only get EmberZNet 7.3.3.x security updates. While the EmberZNet 7.4.x.x will continue to get new bug fix releases.

@estimadarocha FYI, you can also get newer unofficial firmware releases here:

https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds

xsp1989, 2024-03-19:
Personally, I like to use the old version. The new version usually has new functions, and the new functions usually bring new bugs. Then the next few versions are to fix the bugs for this version. Of course, it is also possible It is to add new features, fix bugs, and add new bugs at the same time (joke: coding bug).

In terms of usage, a specific version may have a known problem, and you will know how to solve or circumvent the problem, so the problem may not be a problem. If a new problem arises, you may need to reanalyze the problem in order to solve it, or you can just complain.

sstidl, 2024-05-03:
A reason for updating is that they drop support for older versions in zigbee2mqtt

https://github.com/darkxst/silabs-firmware-builder/issues/84#issuecomment-2090962854

darkxst, 2024-05-05:
> A reason for updating is that they drop support for older versions in zigbee2mqtt

I've not seen anything that they have dropped `ezsp` backend, however the new `ember` backend is currently only available with 7.4.x firmwares.

jonny190, 2024-05-05:
I can confirm @sstidl's comment from my logs
`
[2024-05-05 17:55:49] warning:  zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`

after chaing and attempting start i get

```
[2024-05-05 18:17:33] info:     z2m: Starting Zigbee2MQTT version 1.37.0 (commit #46f34c8)
[2024-05-05 18:17:33] info:     z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-05 18:17:34] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-05-05 18:17:34] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-05-05 18:17:34] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-05 18:17:34] info:     zh:ember:uart:ash: Socket ready
[2024-05-05 18:17:34] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-05-05 18:17:35] info:     zh:ember:uart:ash: ======== ASH connected ========
[2024-05-05 18:17:35] info:     zh:ember:uart:ash: ======== ASH started ========
[2024-05-05 18:17:35] info:     zh:ember:ezsp: ======== EZSP started ========
[2024-05-05 18:17:35] error:    z2m: Error while starting zigbee-herdsman
[2024-05-05 18:17:35] error:    z2m: Failed to start zigbee
[2024-05-05 18:17:35] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-05 18:17:35] error:    z2m: Exiting...
[2024-05-05 18:17:35] error:    z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1651:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:802:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
```

sstidl, 2024-05-05:
I did manage to switch to Tims firmware:
https://github.com/darkxst/silabs-firmware-builder/issues/84#issuecomment-2094078288

jonny190, 2024-05-05:
Ahhhh @sstidl i'm using a one of these https://templates.blakadder.com/ewelink_ZB-GW03.html not sure how i'd go about flashing it

raptr55, 2024-05-05:
> Ahhhh @sstidl i'm using a one of these https://templates.blakadder.com/ewelink_ZB-GW03.html not sure how i'd go about flashing it

Same here. I got this one from AliExpress
https://a.aliexpress.com/_EGwHi11

Not sure about flashing procedure and which new firmware I have to choose to fix the error

darkxst, 2024-05-06:
> using a one of these https://templates.blakadder.com/ewelink_ZB-GW03.html

I believe this requires signed firmware, so I am unable to include in my firmware builder.

xsp1989, 2024-05-06:
I built a 7.4.3.0 firmware today. There are still some problems. If there are no problems, I will release it as soon as possible.

xsp1989, 2024-05-06:
@jonny190 If you purchased the tasmota version, you need to use the signature firmware to refresh the zigbee firmware. If it is the ewelink version, you need to use the serial port to flash it. For details, please see https://thehelpfulidiot.com/a-wired-sonoff-zigbee-alternative

Hedda, 2024-05-06:
> > A reason for updating is that they drop support for older versions in zigbee2mqtt
> 
> I've not seen anything that they have dropped `ezsp` backend

Z2M have not dropped the support for it but they marked `ezsp` adapter as deprecated, see -> https://github.com/Koenkk/zigbee-herdsman/pull/1039

Deprecated means it will no longer maintained and that users now have a grace period to migrate to the new `ember `adapter.

Their new `ember` adapter is suppose to fully replace old `ezsp` adapter but it does require EmberZNet 7.3.0.0 or later firmware.

* https://github.com/Koenkk/zigbee-herdsman/pull/918
* https://github.com/Koenkk/zigbee2mqtt/discussions/21462

xsp1989, 2024-05-28:
Firmware 7.4.3.0 has been updated, signed and unsigned versions have been released

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware

djrm05, 2024-05-28:
> Firmware 7.4.3.0 has been updated, signed and unsigned versions have been released
> 
> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware

@xsp1989 thanks for that. I've tried to upgrade the firmware for a GW03, however it prompt invalid signature. Is there any specific process to upgrade this firmware in this device? Any info would be appreciated. Thanks

xsp1989, 2024-05-28:
> Hi, thank you for updated this. Any chance for router firmware get update? I'm using it with https://github.com/digiblur/Tasmota/tree/development/zigbee_router

There are no plans to update the router at the moment. For the router, updating the firmware will not gain any new features.

xsp1989, 2024-05-28:
> @xsp1989 thanks for that. I've tried to upgrade the firmware for a GW03, however it prompt invalid signature. Is there any specific process to upgrade this firmware in this device? Any info would be [...]

If you are connecting to Z2M or ZHA through TCP, you need to switch the IO configuration to Zigbee RX and Zigbee TX.

djrm05, 2024-05-28:
make sense, but would it be possible to revert the IO configuration later
to TCP to be used as before or would this cause any issue?

djrm05, 2024-05-28:
Thanks but after switching IO config to Zigbee TX/RX I got the same
outcome....invalid signature :(

xsp1989, 2024-05-29:
> […](#)

I have tested it and it can be updated normally. ZB-GW03 should use signed firmware. If it still doesn't work, you can try to restore Tasmota to factory settings and then upgrade.

xsp1989, 2024-05-29:
> > As I understand, you need to have upgraded to at least EmberZNet 7.4.x firmware version or later to use the new ”ember” driver/adapter in Zigbee2MQTT, but from what I read it is suggested to first upgrade to EmberZNet 7.3.x firmware version while still using the old ”ezsp” driver/adapter as it sounds like a migration to the new ”ember” driver will go smoother if already have had the old ”ezsp” driver/adapter working EmberZNet 7.3.x firmware version while still using the old ”ezsp” driver/adapter in Zigbee2MQTT.
> 
> @xsp1989 any chance you can also release a 7.3.x version to make migrating smoother for Zigbee2MQTT users? They are recommending to first upgrade to a 7.3.x version before upgrading to 7.4.x version as otherwise backup and restore will not fully work

The unsigned firmware was released as 7.3.1.0. Today I updated the signed version for ZB-GW03.

robvanoostenrijk, 2024-05-29:
> I have tested it and it can be updated normally. ZB-GW03 should use signed firmware. If it still doesn't work, you can try to restore Tasmota to factory settings and then upgrade.

Experiencing the same problem, with a clean Tasmota installation.

[image]

[image]

Any specific steps which I might have missed?

brunohoff, 2024-05-29:
@robvanoostenrijk, I was able to upload the firmware on my https://templates.blakadder.com/ewelink_ZB-GW03.html following the steps below:

1 - Make a configuration backup
2 - Reset Tasmota configuration
3 - Upload the new firmware: https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota
4 - Restore the backup configuration file

djrm05, 2024-05-29:
> > […](#)
> 
> I have tested it and it can be updated normally. ZB-GW03 should use signed firmware. If it still doesn't work, you can try to restore Tasmota to factory settings and then upgrade.

@xsp1989  Just to confirm, the problem in my case was how I downloaded the ota file as it was 407kB. Just downloaded it again and it is 247,8kB and I upgraded it without reset tasmota configuration and after switch to ember driver, everything was up without issues, so THANKS!
For others with the same issue, pls validate the size of your downloaded file as an additional check.
[image]

robvanoostenrijk, 2024-06-05:
Update for others facing my issue. I was running my ZB-GW03 using the `tasmota-zbbrdgpro` build for a number of months using TCP bridge.

However, turns out it is not the correct Tasmota build with EZSP support built-in, so Tasmota was not able to communicate with the EZSP controller.

I was able to upgrade to the new EZSP firmware successfully after flashing Tasmota `tasmota32-zigbeebridge` from 
https://github.com/tasmota/install/blob/firmware/firmware/unofficial/

    RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 EZSP booted","RestartReason":"Power-on","Code":2}}
    RSL: RESULT = {"ZbState":{"Status":55,"Version":"7.4.3.0","Protocol":13,"Stack":2}}

fuatakgun, 2024-06-17:
@robvanoostenrijk , quick question, I was able to update the bridge into V13 as you pointed out (I had to first enable Zigbee2Tasmota mode as module was disabled in TCP mode and then flash the zigbee firmwares) but I am not able to use the latest version of the bridge with z2m because of many errors. How did you sort it out?

Hedda, 2024-06-18:
> I am not able to use the latest version of the bridge with z2m because of many errors. How did you sort it out?

Such questions about Z2M usage be posted to the Zigbee2MQTT commmunity, see:

https://github.com/Koenkk/zigbee2mqtt/discussions

And

https://github.com/Koenkk/zigbee2mqtt/issues/

fuatakgun, 2024-06-18:
I am coming from z2m repository which asks me to reset nvm from this repository and I failed.

Moreover, I am asking a specific person, let's see.

robvanoostenrijk, 2024-06-25:
> @robvanoostenrijk , quick question, I was able to update the bridge into V13 as you pointed out (I had to first enable Zigbee2Tasmota mode as module was disabled in TCP mode and then flash the [...]

Shortly after I finished the firmware update, I actually switched to a CC2652P2 zstack based coordinator with firmware from https://xzg.xyzroe.cc/. 

The ZB-GW03 is with a friend of mine, he hasn't setup the Ember driver yet. I'll follow up with him for feedback when he does.

mvadu, 2024-07-24:
> @robvanoostenrijk , quick question, I was able to update the bridge into V13 as you pointed out (I had to first enable Zigbee2Tasmota mode as module was disabled in TCP mode and then flash the [...]

what errors are you getting post update? referring to this? https://github.com/Koenkk/zigbee2mqtt/issues/23164

closed 2025-03-04 by xsp1989