xsp1989/zigbeeFirmware #50: Help updating to ncp-uart-sw_6.10.9_115200.ota

danieldabate, 2024-03-25:
Hi! I have a ZB-GW03-V1.4 flashed with ESPHOME and I tried updating its Zigbee firmware to **ncp-uart-sw_6.10.9_115200.ota** without success. 

The update finishes fine, but after reboot neither Z2M nor ZHA are not able to communicate with the device. Flashing **ncp-uart-nsw_6.5.5_115200.ota** and **ncp-uart-sw_6.7.10_115200.ota** work fine, so right now I'm using the last one (seems to be the same version that it came with from the factory).

I'm not sure how to debug this or if there is anything I'm missing. Any guidance is welcome!

xsp1989, 2024-03-26:
I have updated the firmware, please try the new firmware. If you have any questions, please give me feedback.

danieldabate, 2024-03-26:
It worked now!! Thanks for your help. Just out of curiosity, what was wrong with the other file?

closed 2024-07-26 by xsp1989