xsp1989/zigbeeFirmware #51: Updating firmware

brandonb1987, 2024-07-23:
Is there a method to flash the firmware without using Telnet? Every time and method (OS) I've used just continues to disconnect and doesn't end up uploading the new firmware. I'd prefer to do it over serial if possible. 

Is this possible? Or can you point me to something that shows how? Because at this point I'm stuck. I have ESPhome working with 6.7.10 but I am not able to update the firmware no matter what I try.

brandonb1987, 2024-07-25:
Tried the Tasmota route before posting this. Just continues to fail.

xsp1989, 2024-07-26:
There will be no problem upgrading directly through the tasmota upgrade page. You can try downloading the upgrade file again. Maybe an error occurred when downloading the file, causing the upgrade to fail.

brandonb1987, 2024-07-28:
Just tried other versions, and redownloaded every file. Still get the same thing when trying to upgrade the zigbee firmware.

The last one ended with 
392/49K   Retry 0: Cancelled

I don't know why this is so. The only thing I can think of is its constantly reseting the Telnet? I don't know why it would be doing that. The esphome code and ZHA integration works totally fine. I just can't do the Telnet.

Which is why I'm curious if there is software or some sort of guide that can show me how to upload the file by serial and not Telnet. If I can just do it hardwired and by serial, then this shouldn't be an issue anymore. Theoretically.

mvadu, 2024-07-28:
If you use  Tasmota (from @xsp1989 for gw-03) , it will do serial communication with zigbee microcontroller. You can't stay in Esphome and say you have tried tasmota route.

brandonb1987, 2024-07-29:
Ok...let me spell this out...

I was originally on 14.0 Tasmota. I upgraded the zigbee firmware to 6.7.10. Worked fine. Decided to try out ESPhome with ZHA instead so I don't have to do the yaml all the time. Settled on ESPhome so I tried doing the zigbee firmware upgrade to 7.4.3. Kept getting the error above.

So the first thing I did before opening this thread was go back onto Tasmota, do the firmware upgrade, with the intent of going back onto ESPhome once the firmware was upgraded. IT DON'T WORK...

Hence why I've posted on here with the question if its possibly to do a serial upgrade while on ESPhome. As in, give me some way to do it in a way that is hardwired and not reliant on a remote connection...

I'm not aware of a way to do it over serial with Tasmota either as I've tried doing it the normal route of just simply using the built in firmware upgrade option tab, upgrading the zigbee firmware using the .ota file and being done with it. Which has worked in the past.

So hopefully instead of rehashing things I've already done and tried, someone can see I've tried it all and it still refuses to upgrade, and even downgrade...

Constructive and helpful solutions needed, not telling me to do something I've already done....

xtypebox, 2024-09-12:
> Try this.. [#49 (comment)](https://github.com/xsp1989/zigbeeFirmware/issues/49#issuecomment-2137189283)
> 
> I was able to upgrade to ember firmware without having to repair any of the devices. Using https://github.com/xsp1989/ZB-GW03_firmware/tree/master/tasmota/v14.0.0/firmware as main tasmota firmware.

Can you explain how you do it without having to repair?

Hedda, 2025-01-20:
I think this can be closed since not a firmware issue.

FritzTheCatfish, 2025-02-06:
On my Sonoff ZBBridge using Tasmota 14.4.1 I am too unable to load the router sw.
I have tried all the recent versione via OTA with the same results, it starts but never ends.
If I try via local file update I get the signed error.
If I try with the 2022 version, it starts but never ends.
Any suggestione?
Is there an issue with tasmota 14.4.1?
Thanks in advance

closed 2025-03-04 by xsp1989