xsp1989/zigbeeFirmware #52: EmberZNet 7.4.4.0 and 8.0.2.0 Zigbee Coordinator NCP firmware images

cfilipem, 2024-10-27:
@xsp1989 could you compile Zigbee EmberZNet 7.4.4.0 and 8.0.2.0 Zigbee Coordinator NCP firmware images for these adapters?

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.4.4.0.pdf

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-8.0.2.0.pdf

xsp1989, 2024-10-28:
There is no plan to build 8.x firmware for the time being. Generally, new cross-version upgrades will have some major changes, and often introduce some bugs. Unless there are other features that must be supported by the new version, it will be considered for release.

Hedda, 2025-01-20:
For reference, EmberZNet 8.1.0 firmware is buggy (software flow control does not seem to work in EmberZNet 8.1.0 firmware), so better to stick with 8.0.2 and newer 8.0.x releases for now if want to go with 8.x.x

gpaesano, 2025-03-29:
I support  a signed version of EmberZNet 7.5.0.0 (7.5.0 is available in SDK 4.4.6) would also be nice since has fixes that previous 7.4.x.x releases did not have.