xsp1989/zigbeeFirmware #53: [ZigBee FW] ncp-uart-hw-v7.4.5.0 is released. Signed version for ZB-GW03 wanted

ESP4Ever, 2024-12-19:


derzahla, 2025-01-09:
@xsp1989 

I am experiencing issues as well, described here: https://github.com/Koenkk/zigbee2mqtt/discussions/23224

I have a ZB-GW03 v1.4.  with your 7.4.3 fw.  Would be great to have some newer builds to try out.  Or can you point me to instructions on how to build myself?  I dont suppose ZB-GW04 firmware would work for me, or some of the others, would it?

farmerkz, 2025-01-13:
I use several coordinators based on EFR32: ZB-GW03, ZB-GW04, Sonoff E.
Only on ZB-GW03 I use version 7.4.3, on all the others I use version 7.4.5, and on the test coordinator 8.0.2. Unfortunately version 7.4.3 loses to newer versions in stability, besides it has many errors like "network/route error".
@xsp1989,  can I know if there is a chance that firmware or version 7.4.5 or 8.0.2 will be compiled for it in the near future?
Thanks!

derzahla, 2025-01-14:
@farmerkz thanks for the reply. Im fairly new to zigbee. Based on things I read last year, I bought the zb-gw03 wifi/eth(I only use wired eth) board as a coordinator and bought two usb dongles that I now realize have zb-gw04 chips.  For some reason I dont recall, from what I had read at the time I was under the impression that the zb-gw03 model was the best choice for a coordinator using z2m and that the usb zb-gw04 dongles made good routers but not coordinators.  

After setting up zigbee devices throughout my home I realized that everything with a hardwired power source acted as a router itself and saw no need for the usb dongle routers anymore, so here they sit.

Would I be better off using the gw04 dongles over my zb-gw03 tasmota eth-zigbee bridge?

Mik145, 2025-01-15:
**@xsp1989**
It would be great if you made a new firmware. Thank you.

xsp1989, 2025-01-16:
I have made firmware 7.4.5.0, including signed firmware and unsigned firmware

farmerkz, 2025-01-16:
@xsp1989 
Excellent, thank you!!!

farmerkz, 2025-01-16:
Successfully updated the firmware, no problems found so far.
@xsp1989 
Thanks again :)

derzahla, 2025-01-16:
@xsp1989 Thank you Sir!

derzahla, 2025-01-16:
@farmerkz @xsp1989 this is odd.  i keep getting failed: invalid file signature when i try to update with https://github.com/xsp1989/zigbeeFirmware/raw/refs/heads/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw-7.4.5.0_115200.ota. Ive tried about 20 times now, from tasmota, from safeboot, using the link, uploading the file from my laptop. Ive tried updating safeboot, upgrading to the latest "official" unofficial/tasmota32-zigbeebridge.bin. Holy hell, any idea what I am doing wrong?  I kinda remember some similar issues last time I updated the zigbee firmware, but it was quite awhile ago and I dont remember how I got it to work.

ESP4Ever, 2025-01-17:
It is almost one day since FW update I see no z2m restarts nor errors in the log. Thanks again, one headache less)))

farmerkz, 2025-01-17:
@derzahla I make a backup copy of the configuration, reset the configuration and update the firmware from a previously downloaded file, then restore the configuration from the backup copy.

derzahla, 2025-01-17:
@farmerkz thanks for the tips. I finally got it. For anyone else having trouble -  The steps that finally got it working for me(not sure if they were all necessary - note I was on the latest unofficial/tasmota32-zigbeebridge.bin fw at the time ) were:

1) backup my config(after reset still apply zigbee fw update)
2) applied the zigbee2tasmota configuration template:

```
{"NAME":"ZB-GW03-V1.2","GPIO":[0,0,3552,0,3584,0,0,0,5793,5792,320,544,5536,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,608,640,32,0,0,0,0,0],"FLAG":0,"BASE":1}
```
Still no-go, ncp-uart-sw-7.4.5.0_115200.ota would 'upload' and 'transfer' but then nothing.  If flashed from safe-boot it would still fail on signature.

3) Flashed back to  xsp1989's v14.2.0/firmware/tasmota32-ZB-GW03-EN.bin fw.  Now when I uploaded ncp-uart-sw-7.4.5.0_115200.ota from the webui, it "uploaded", "transfered" and then rebooted into safeboot to complete the update!

I could then restore my original config with uses the below template, which as I understand it, simply enables tcp->serial access to the zigbee radio, no zigbee2tasmota stuff.

```
{"NAME":"Zig_Bridge32","GPIO":[1,1,3552,1,3584,1,1,1,5793,5792,320,544,5536,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,608,640,32,1,1,0,0,1],"FLAG":0,"BASE":1}
```

SUCCESS! thank guys. 

Note: If anyone could help elucidate some of the following for me I would greatly appreciate it!  

1) I am still pretty confused about why I needed to change my config template and/or return to xsp1989's v14.2.0/firmware/tasmota32-ZB-GW03-EN.bin fw to get the zigbee radio to flash. 

2) Also do not understand the details of the signature requirements and why I would get 'file signature failed' before. As I understand it, the bootloader requires the signature and none of the firmware updates are touching the bootloader so why the difference? 

3) Furthermore, who has the key needed to sign the updates? Just @xsp1989? If so could @xsp1989 provide a bootloader update that doesnt require signing? That way we could presumably build our own updated firmware instead of having to bother @xsp1989 to do it

Thanks again @xsp1989 for the update

PsychoRS, 2025-01-21:
Thanks @xsp1989 , but... is it possible to get this signed firmware with tokens interface, as stated by @Nerivec, author of ember-zli command line interface, here https://github.com/Nerivec/ember-zli/issues/15? I need to make a NVM3 tokens backup of my ZB-GW03 to migrate to an SLZB-06M.

Thanks in advance.

kellerman, 2025-02-19:
> 3\. rebooted into safeboot to complete the update. SUCCESS!

@derzahla 

With this template:
`{"NAME":"ZB-GW03-V1.2","GPIO":[0,0,3552,0,3584,0,0,0,5793,5792,320,544,5536,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,608,640,32,0,0,0,0,0],"FLAG":0,"BASE":1}`
You can flash the ZB firmware also with the lastest development branch of Tasmota. I struggled finding this too 😄 , but yeah, after that just change Zigbee RX and TX to TCP RX and TX if you use the usb2ser. With latest FW 7.4.5.0 from @xsp1989 the ember protocol works with Zigbee2MQTT just fine!
Thanks for all your replies, they helped!

Mik145, 2025-02-19:
@xsp1989 
Thank you very much for your work. 
Everything works great.

closed 2025-03-04 by xsp1989