xsp1989/zigbeeFirmware #57: Possible to use Silicon Labs Flasher Add-on for ZB-GW03?

ElVit, 2025-03-03:
Hi,

I don't know if this question was already asked, but I didn't found it here.
Is it possible to use a Silicon Labs Flasher Add-on to flash ZB-GW03?

As far as I know there are 3 Add-ons available:
1. [Official Silicon Labs Flasher](https://github.com/home-assistant/addons/tree/master/silabs_flasher)
2. [TubesZB Silicon Labs FW Flasher](https://github.com/tube0013/tubeszb_addons/tree/main/tzb-silabs-flasher)
3. [ZigStar Silicon Labs FW Flasher](https://github.com/mercenaruss/zigstar_addons/tree/main/zigstar-silabs-flasher)

Option 1 will probably not work because as far as I can see it supports only usb devices.
Option 2 and 3 are quite similar. Both supporting also network devices.

All 3 options requiring a GBL file. 
But if I understand it correct (from the other issues here) an OTA file is actually the same as a GBL file, with the only difference being that the OTA file is signed.

I use the ZB-GW03 with ESPHome.
Setup was made with the flashing instructions from syssi:
- https://github.com/syssi/esphome-zb-gw03/blob/main/docs/flashing.md

If I start flashing with the ZigStar addon I see the follwing output:
```
[14:11:09] INFO: Starting universal-silabs-flasher with socket://192.168.178.25:6638 (bootloader baudrate 115200)
2025-03-03 14:11:12 193f5383-zigstar-silabs-flasher universal_silabs_flasher.flasher[148] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-03-03 14:11:12 193f5383-zigstar-silabs-flasher universal_silabs_flasher.flasher[148] WARNING No application can be launched
2025-03-03 14:11:12 193f5383-zigstar-silabs-flasher universal_silabs_flasher.flasher[148] INFO Detected bootloader version '1.9.1.4'
2025-03-03 14:11:12 193f5383-zigstar-silabs-flasher universal_silabs_flasher.flasher[148] INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.9.1.4' at 115200 baudrate (bootloader baudrate 115200)
firmware.gbl
```
Then nothing is happening anymore.
Can you tell me if it is simply not possible to flash an OTA file this way? Or am I doing something wrong?

xsp1989, 2025-03-04:
ZB-GW03 can only use signed firmware, unless you use jlink V9 to refresh the bootloader

ElVit, 2025-03-04:
If I follow the instruction of syssi (see my comment above) I can flash the ota file (ncp-uart-sw_7.4.3.0_115200.ota) from your repo successfully.
So do I understand you correct that it is simply not possible with the add-on?

Syssi's manual is using the tool [lrzsz](https://www.ohse.de/uwe/software/lrzsz.html) which uses the XMODEM protocol to flash the zigbee module.
The addons using the [NabuCasa's Universal Silicon Labs Flasher](https://github.com/NabuCasa/universal-silabs-flasher) which is also using the XMODEM protocol.
I'm not familiar with the XMODEM protocol, but shouldn't the flash process be similar if they use the same protocol?

xsp1989, 2025-03-07:
The ZB-GW03 bootloader has been slightly modified, I don't know if it is compatible with other flash tools, but it does use the xmodel protocol, which has not been modified. But in any case, only the signed firmware I released can be used.

closed 2025-03-11 by xsp1989